RIOT-OS/RIOT #5241: Outdated M3 documentation on the wiki page

kYc0o, 2016-04-04:
Just a reminder to update IoTLab M3 node on the wiki page

https://github.com/RIOT-OS/RIOT/wiki/Board%3A-IoT-LAB-M3

aabadie, 2016-04-04:
Here are (minor) requested changes (I still can't push to the wiki repository): 
```
diff --git a/Board:-IoT-LAB-M3.md b/Board:-IoT-LAB-M3.md
index 555aae6..64f720d 100644
--- a/Board:-IoT-LAB-M3.md
+++ b/Board:-IoT-LAB-M3.md
@@ -35,11 +35,11 @@
 |              | RNG           | no            | no HW module |
 |              | Timer         | full          ||
 | Radio Chip   | AT86RF231     | partly                | will be remodelled soon |
-| Accelerometer        | L3G4200D      | yes           | planned |
-| Magnetometer | L3G4200D      | yes           | planned |
-| Gyroscope    | LSM303DLHC    | yes           | planned |
-| Pressure Sensor      | LPS331AP      | yes           | planned |
-| Light Sensor | ISL29020      | yes           | planned |
+| Accelerometer        | L3G4200D      | yes           |  |^M
+| Magnetometer | L3G4200D      | yes           |  |^M
+| Gyroscope    | LSM303DLHC    | yes           |  |^M
+| Pressure Sensor      | LPS331AP      | yes           |  |^M
+| Light Sensor | ISL29020      | yes           |  |^M
 
 ##### Note on at86rf231 radio driver
```

haukepetersen, 2016-04-06:
updated the wiki page. @kYc0o: please check if you find this sufficient...

@aabadie: Everyone should be able to push to the wiki, did you try editing it using the web interface directly, too?

OlegHahm, 2016-04-06:
> @aabadie: Everyone should be able to push to the wiki, did you try editing it using the web interface directly, too?

I'm not sure about the Git access to the Wiki, but indeed, everyone should be able to *edit* the wiki using the web interface. (Just checked with another account.)

aabadie, 2016-04-06:
@OlegHahm, @haukepetersen, I can indeed edit using the web interface. Thanks !

kYc0o, 2016-04-06:
@haukepetersen Yeah those changes should be enough, @aabadie maybe you can fix the images that are not shown?

aabadie, 2016-04-06:
@kYc0o : fixed. There were outdated links to the `devgrenoble.senslab.info` host. I replaced them with  `www.iot-lab.info`

haukepetersen, 2016-04-08:
@aabadie, @kYc0o: are you happy with the updated board wiki page for now so we can close this issue?

aabadie, 2016-04-08:
Yes !

kYc0o, 2016-04-08:
Yes it's OK for me too

closed 2016-04-08 by OlegHahm