RIOT-OS/RIOT #5257: at86rf2xx: receiving own broadcast?

OlegHahm, 2016-04-06:
While debugging #4725 I observed packets being broadcasted over the wireless interface that were received by the sending host itself on the same interface. Maybe a bug in our driver?

OlegHahm, 2016-04-07:
I think I found the cause:
2 (or more packets) sent directly back to back will cause the driver to block in `at86rf2xx_tx_prepare()`. Once the first packet is sent out, the thread will resume, and finally handle the pending `netdev2->isr()` calls. The first call will reset state to `AT86RF2XX_STATE_RX_AACK_ON` causing the following ones to handle the IRQ as an RX interrupt.

haukepetersen, 2016-04-08:
Following our offline discussion about this problem, I had the following idea this morning (just thinking loud here, so it may make no sense at all):

Problem: we use synchronization on radio states, as the thread running the driver might process transmit commands coming from upper layers before interrupt handle requests, right?! This might lead to TX_END events to be miss-interpreted as RX_COMPLETE, leading to the observed behavior. So the problem is, that we need to wait in `_tx_prepare()` until the current TX is done, without blocking the thread, as the thread needs to act on a message that is received on the TX_END event. So as a intermediate fix, how about this:
_tx_prepare():
 - is tx ongoing?
    - no: proceed as usual
    - yes: send GNRC_NETAPI_MSG_TYPE_SND message to own PID

This will result in a loop, where the SND message is put to the back of the message queue, as long as the transceiver is processing the current packet. This results in kind of an active waiting, but this is not different then the current situation with the `while` loop in the _tx_prepare() function. Also the order of packet transmission should not be changed, since as long as we stay in this 'loop', no other NETAPI_SND packets can be received from the layers above, as the MAC thread has a higher prio and the upper-layer threads are not scheduled in between.

As said, this is far from ideal, but it should solve this issue for now without being more inefficient than the current implementation...

haukepetersen, 2016-04-08:
But #5261 does the job as well, so let's go with that for now.

OlegHahm, 2016-04-08:
Your proposal is similar to the approach in the current `conn` implementation and IIRC in tftp. I'm not a big fan of this solution as it is a bit hacky and kind of misusing the IPC. However, I agree, it would work and now that I think about it, it might be even the better fix in case of small message queues. (What currently happens if continuous send calls, fill up the message queue? Probably interrupts get lost.)

The IMO conceptional better solution would be to have some kind of prioritized (or at least distinct) message queues (or mailboxes). The disadvantage that I see for this solution is the memory overhead. (We could go for thread_flags, but at least GNRC will still need a message queue _and_ thread_flags.)

haukepetersen, 2016-04-08:
> it might be even the better fix in case of small message queues. (What currently happens if continuous send calls, fill up the message queue? Probably interrupts get lost.)

Actually I am not sure if the message queues will fill up currently -> in `at86rf2xx_tx_prepare()` we have an active waiting loop, so the thread actively blocks, preventing upper layer threads to be scheduled and putting new messages into the queue, right?

But anyway, lets aim for using thread_flags as better solution for the mid-term, and use #5261 as quick-fix for the release.

OlegHahm, 2016-04-08:
As I said: I'm not so sure that thread_flags are a better solution. (Yes, they are better from a conceptional view, but not necessarily concerning the memory usage.)

OlegHahm, 2016-04-08:
Fixed by #5261.

closed 2016-04-08 by OlegHahm