RIOT-OS/RIOT #5259: at86rf2xx: idle_state not set correctly

OlegHahm, 2016-04-07:
While analyzing #5257, I observed that `at86rf2xx_setup()` initially set `idle_state` to `AT86RF2XX_STATE_TRX_OFF`. This is only changed in `at86rf2xx_tx_prepare()` where it is set to either `AT86RF2XX_STATE_BUSY_RX_AACK` or `AT86RF2XX_STATE_BUSY_TX_ARET`. This means that `idle_state` will change after the first sent (loaded) packet and never go back to this initial value.

I think `idle_state` is not used anywhere else, so I would vote to simply drop it, or do I miss something?

OlegHahm, 2016-04-20:
After discussing it briefly with @haukepetersen I think it is more about an ill-named variable than an actual bug.

miri64, 2016-10-18:
Then I'll add it as a fixable issue to the current release.

OlegHahm, 2017-01-15:
Invalid.

closed 2017-01-15 by OlegHahm