RIOT-OS/RIOT #5263: netdev: use thread_flags

OlegHahm, 2016-04-08:
#5261 introduced a counter for pending TX calls. A conceptional better solution would be to use `thread_flags` instead of IPC for netdev devices. This would allow to prioritize IRQ events over other calls like send and hence guarantee the correct order of execution.

kaspar030, 2016-04-08:
That's basically what thread_flags are designed for: safe flagging of events, even from ISRs.

With gnrc, we would need to beef up the messaging to set a thread flag, otherwise, currently, a thread can only wait for either a flag or a message.

kYc0o, 2016-07-26:
Let's try to propose a solution for this for the next release.

miri64, 2016-10-31:
If this issue is addressed this might fix #5486 too ;)

OlegHahm, 2017-01-15:
I don't think that we get this implemented before the release.

PeterKietzmann, 2019-08-19:
@jia200x I'm mentioning you because the issue relates to your work and I don't want it to be closed automatically

jia200x, 2020-02-20:
This should still be open

jia200x, 2020-09-01:
Since https://github.com/RIOT-OS/RIOT/pull/14791 got merged, I think we should really stop using `msg_t` for processing IRQs.
If a message gets lost from IRQ context, the `dev->driver->isr()` function will never be called and the driver would be stuck forever.
This happens quite often with `tests/driver_at86rf2xx` under stress conditions.

As a corollary if we assume that all network stacks would use events (`event_t`), the initialization and bottom half processing can be leveraged to the OS instead of the network stack.

If the `xxx_setup` function from the driver accepts an IRQ callback and the radio exposes a `xxx_irq_handler()` to handle the IRQ request, one could simply do on auto_init:

```c
typedef struct {
    event_t event;
    at86rf2xx_t dev;
} at86rf2xx_bh_t;

static at86rf2xx_bh_t at86rf2xx_devs[AT86RF2XX_NUM];

static void _isr_at86rf2xx(void *arg)
{
    at86rf2xx_bh_t *bh = container_of(arg, at86rf2xx_bh_t, dev);
    int index = bh - at86rf2xx_devs;
    event_post(AN_EVENT_QUEUE, &bh->event);
}

/* This is our Bottom-Half processor */
static void _irq_event_handler(event_t *event)
{
    at86rf2xx_bh_t *bh = (at86rf2xx_bh_t*) event;
    at86rf2xx_irq_handler(&bh->dev);
}

void auto_init_at86rf2xx(void)
{
    for (unsigned i = 0; i < AT86RF2XX_NUM; i++) {
        LOG_DEBUG("[auto_init_netif] initializing at86rf2xx #%u\n", i);
        at86rf2xx_setup&at86rf2xx_devs[i].dev, &at86rf2xx_params[i], _isr_at86rf2xx); /* Note we assign the IRQ to our function that posts events */
...

Note that:
- The initialization code is unified for all network stacks. No need to add device specific code in the initialization of network stacks (see e.g https://github.com/RIOT-OS/RIOT/blob/master/pkg/lwip/contrib/lwip.c#L81)
- This doesn't prevent processing the radio differently, since it's always possible to change the IRQ callback. A network stack could simply do:
```
static void _isr_at86rf2xx(void *arg)
{
    /* Call IRQ handler from ISR (for specific use cases) */
    at86rf2xx_irq_handler(arg);
}

void auto_init_at86rf2xx(void)
{
    for (unsigned i = 0; i < AT86RF2XX_NUM; i++) {
        LOG_DEBUG("[auto_init_netif] initializing at86rf2xx #%u\n", i);
        at86rf2xx_setup&at86rf2xx_devs[i].dev, &at86rf2xx_params[i], _isr_at86rf2xx); /* Note we assign the IRQ to our function that posts events */
```
- The network stack only needs to care about defining the `event_callback` and give context about `AN_EVENT_QUEUE`. For GNRC this could mean `netif->evq`, but other network stacks could simply use the `event_thread` module.

Does this sounds reasonable?

@maribu, @kaspar030, @bergzand and @benpicco might also be interested

closed 2021-06-03 by stale[bot]