RIOT-OS/RIOT #5265: periph/timer: Remove timer_irq_enable, timer_irq_disable?

gebart, 2016-04-08:
AFAICS timer_irq_enable, and timer_irq_disable are only ever called from within the timer periph implementations themselves. Do we really need to keep them in the public API?
What is the use case for disabling the IRQ of a timer without stopping it?

kYc0o, 2016-07-26:
Is this still an issue? If so, let's try to address it for the next release.

gebart, 2016-07-26:
yes, still an issue. This thread has been awfully silent since I posted it.

OlegHahm, 2016-07-26:
Oops, yes, this one slipped through somehow. Could it be that the old hwtimer module needed this API somehow? For now, I would support this RFC.

haukepetersen, 2016-10-17:
the functions were used from the old `hwtimer`. Since that code path does not exist anymore, I would be happy if we would just remove those functions.

miri64, 2016-11-01:
I just closed #5979 since this functions are still very entangled on the boards that implement them, so I postpone.

smlng, 2017-03-01:
fixed by #6360

closed 2017-03-01 by smlng