RIOT-OS/RIOT #5282: Timeout for I2C, SPI, UART

sdalu, 2016-04-09:
It would be nice to have a timeout while reading/writing on the following peripherals, like what is done in ChibiOS

gebart, 2016-04-10:
- I2C: Agreed, clock stretching or misbehaving slave devices can potentially make a transaction stall, depending on the periph implementation this may hang indefinitely.
 - SPI: No, SPI only needs a timeout in slave mode which is not implemented right now (the master drives the clock, time to complete is always deterministic)
 - UART: the periph/uart API only supports RX via asynchronous callbacks per byte received, and UART TX is driven by the MCU clock, so the TX time is always deterministic.

sdalu, 2016-04-12:
- SPI: I guess slave mode is going to be implemented some days, if it will require changes to the existing SPI API to handle timeout, perhaps it would be better to prepare for it earlier and minimize futur code change. 
- UART: if flow control is enabled this could block TX

PeterKietzmann, 2016-05-17:
Hi. Can we merge this issue and #3064 somehow and close one? I tend to paste your outcomes into  #3064. Or does the issue address a different problem?

sdalu, 2016-05-18:
Same issue.

PeterKietzmann, 2016-05-19:
I posted your comments in #3064. Please continue discussions in there. Will close this issue now.

closed 2016-05-19 by PeterKietzmann