RIOT-OS/RIOT #5287: feature: implement NETOPT_ACK_REQ

aeneby, 2016-04-10:
Currently, there is a flag for toggling the "Acknowledge Request" bit of the FCF (`NETDEV2_IEEE802154_ACK_REQ`), but no NETOPT for getting/setting it via netdev. Perhaps there should be.

authmillenon, 2016-04-11:
This was originally introduced the way it is right now by @haukepetersen in the original `at86rf2xx` driver so I reassign to him, because I did not know his original intention.

aeneby, 2016-04-12:
See #5297

haukepetersen, 2016-04-13:
actually, this was intended to be covered by `AUTOACK` if I remember correctly... So what is the diff between `NETOPT_AUTOACK` and `NETOPT_ACK_REQ` now?

aeneby, 2016-04-13:
I think you're misinterpreting `AUTOACK`, which is not the same thing as `ACK_REQ` (IMO [this define](https://github.com/RIOT-OS/RIOT/blob/6c817c888ea8ba93618e2eca736d29c88a042453/drivers/include/at86rf2xx.h#L119) is incorrect, for example). `AUTOACK` should tell the driver to automatically acknowledge frames which have the "acknowledge request" (AR) bit set in the FCF, whereas `ACK_REQ` should be used to set the AR bit in outgoing frames. We need two options to handle these two cases, hence the patch.

aeneby, 2016-04-26:
@haukepetersen any feedback on the referenced PR?

authmillenon, 2016-04-26:
He is on holiday now for a while. I fear this has to wait for now.

authmillenon, 2016-04-26:
(maybe if I find some time I can look into it).

aeneby, 2016-04-26:
Okay - thanks, @authmillenon

PeterKietzmann, 2016-04-26:
@thomaseichinger is pretty experienced with that device. Would you mind taking that issue/PR over? In that case, please self-assign as Hauke won't be available for the next months.

PR #5297 (2016-05-31): Implement NETOPT_ACK_REQ configuration option.
This fixes #5287 - currently the option `NETOPT_AUTOACK` is incorrectly being used to set the AR (acknowledge request) bit of the 802.15.4 MAC header. This PR introduces a new option for doing this instead, `NETOPT_ACK_REQ`, leaving it up to the driver to set AUTOACK when requested.
closed 2016-05-31 by PeterKietzmann