RIOT-OS/RIOT #5296: core: create message queue on thread creation

OlegHahm, 2016-04-11:
Currently, a thread has to create its own message queue. This could lead to an ugly situation, where a thread is created (via `thread_create()`) with a lower priority and other threads (all with higher priorities) start already sending IPC messages to the new thread. Even if the thread would call `msg_init_queue()`, this won't be called in time and messages will get lost (if non-blocking IPC calls are used).

There are two ways to go about this:
1.) Leave it as is and consider this a design decision. This would, for instance, make it impossible to create a GNRC thread with a lower priority and register it after creation from the creating thread.
2.) Add a parameter to `thread_create()` to pass a message queue right on thread creation. (Or provide a function to create message queues for other threads.)

Thanks @tfar for reporting this issue.

OlegHahm, 2016-04-11:
Actually, I think adding a parameter to `thread_create()` would be the most elegant solution, but introduce a lot of changes.

kaspar030, 2016-04-11:
I would definitely keep it as is.

- non-blocking IPC is *expected* to drop messages unless *very* carefully used
- one thread registering another is extremely bad design
- if absolutely needed for threads that might not have created their message queue yet, the assert on netreg should be turned into a develhelp warning

kaspar030, 2016-04-11:
We're currently using 8 thread creating flag bits. We *could* use like 8 bits of that for ```INIT_MSG_QUEUE_SIZE_1 ... 256``` and then, if set, allocate and set up the queue in thread_create(). API-wise it would be a minor change, and it probably even saves some bytes in .text.

OlegHahm, 2016-04-12:
> one thread registering another is extremely bad design

Can you elaborate on this?

miri64, 2016-10-18:
@kaspar030 ping?

kaspar030, 2016-11-18:
>> one thread registering another is extremely bad design
> Can you elaborate on this?

Maybe "extremely bad design" is exaggerated, but a thread gets "registered" for something by another thread, the thread has no control over when it get's registered, and it might not be ready. It might even have preconditions that are not met and decide not to register at all. I can't quote any book here, but IMHO having a thread deal with it's own stuff makes the overall design much cleaner.

I think the example you've given in the PR description shows such a problem that could have been avoided:

"This could lead to an ugly situation, where a thread is created (via thread_create()) with a lower priority and other threads (all with higher priorities) start already sending IPC messages to the new thread. Even if the thread would call msg_init_queue(), this won't be called in time and messages will get lost (if non-blocking IPC calls are used)."

(I assume that other threads only start sending IPC messages *after* the thread has been registered for receiving them.)

OlegHahm, 2016-11-18:
> but IMHO having a thread deal with it's own stuff makes the overall design much cleaner.

If we follow this paradigm, shouldn't we change the signature of `gnrc_netreg_register()` then to only require the `type` and the `demux_ctx`, but not the PID? This way a thread would only be able to register itself.

kaspar030, 2016-11-18:
> This way a thread would only be able to register itself.

Sounds good to me. @miri64 how often do threads register another?

miri64, 2016-11-18:
I would have to `git grep` that myself... But I see some issues (with both the idea and your statements):

1. I don't see the relation msg-queue as `thread_create()` parameter <-> changes to `gnrc_netreg`... Can you please elaborate how this happened
2. `gnrc_netreg_register()` does not receive a PID as input parameter, but a struct initialized by `gnrc_netreg_entry_init_pid()/GNRC_NETREG_ENTRY_INIT_PID()`.
3. modules like `gnrc_pktdump` (which to my knowledge is the only one where "remote" registration happens atm) wouldn't be possible anymore => the whole in-stack sniffing feature of GNRC would be dropped.

OlegHahm, 2016-11-18:
> I don't see the relation msg-queue as thread_create() parameter <-> changes to gnrc_netreg... Can you please elaborate how this happened

@kaspar030 proposed to consider one thread "managing" another thread's resources, e.g. registering thread A for message reception from thread B, as bad style. This means that `gnrc_netreg_register()` should only be called for the thread itself, i.e. the thread would not need to specify a PID anymore.

> gnrc_netreg_register() does not receive a PID as input parameter, but a struct initialized by gnrc_netreg_entry_init_pid()/GNRC_NETREG_ENTRY_INIT_PID().

Which leads to the same result.

> modules like gnrc_pktdump (which to my knowledge is the only one where "remote" registration happens atm) wouldn't be possible anymore => the whole in-stack sniffing feature of GNRC would be dropped.

Valid point. But couldn't `gnrc_pktdump` provide a function to register itself instead?

(Btw. I'm not yet sure if I agree with @kaspar030's assessment that registering another thread is "bad design".)

closed 2019-09-10 by stale[bot]