RIOT-OS/RIOT #5299: tests: Fletcher test vectors are too short

gebart, 2016-04-12:
https://github.com/RIOT-OS/RIOT/pull/4938#issuecomment-198878721

It might be a good idea for the fletcher tests to have a test input vector which accumulates enough to "wrap around" the modulo part of the algorithm. The current test vector is quite short and containing small numbers.

miri64, 2016-10-18:
Might be fixable for this release.

miri64, 2016-10-20:
I tried to implement it, but it seems that either I'm failing to get a correct value for a long array at http://www.nitrxgen.net/hashgen/ or fletcher32 is broken :-/

miri64, 2016-10-21:
Ah, should have read the comments for the `atof` test ^^ "`XXX: not verified with external implementation yet`" and putting in that data also yields a different hash than the one expected in the test.

PR #5976 (2016-10-26): tests: unittests: add tests for longer input vectors
Fixes #5299
commit 100991e5 (2016-10-26): tests: unittests: add tests for longer input vectors

Fixes #5299
closed 2016-10-26 by miri64