RIOT-OS/RIOT #5302: gpio_init implementaion

malosek, 2016-04-12:
Hello riots!

have been playing with sw i2c for msp430, where do not have open drain outputs, so for 0 have to set pin to output 0 and for 1 configuring pin as input with ext pullups.
For sequence of 0s on i2c it melts down to multiple calls of gpio_init (sda_gpio, GPIO_OUT).
The problem is that gpio_init configures pin as input first and only after as an output with default value of 0.
   /* reset pin and output value */
    port->DIR &= ~(_pin(pin));
    port->OD &= ~(_pin(pin));
    if (mode == GPIO_OUT) {
        port->DIR |= _pin(pin);
    }
that short period when pin is configured as input while pullup is connected creates unwanted peak on the i2c bus. It is not violating i2c spec since it happens while clock is low, but even Im not an EMC pro, do not want short/sharp peaks on long pcb traces as i2c-s are.

Setting gpio as output I would expect that pin will be set to output and not to input and output afterwards. Have seen this approach for other platforms but in general reporting on msp430fxyz.
What is the idea behind? Would a change be acceptable?

btw: If there is interest, I could prepare PR with sw i2c implementation. Or is already in somewhere?

wbr
malo

haukepetersen, 2016-04-14:
The implementation was not really designed for fast, multiple re-initializations as you use it here. But I guess this could simply be fixed with
```c 
/* reset pin and output value */
if (mode == GPIO_OUT) {
    port->DIR |= _pin(pin);
}
else {
    port->DIR &= ~(_pin(pin));
}
port->OD &= ~(_pin(pin));

```
So please go ahead and PR this change!

Also, your soft-I2C would be very interesting! It would be great if you could share it!

PeterKietzmann, 2016-04-26:
@malosek, Hauke proposed a solution where the pin is directly configured as output pin, without resetting the concerning bit before (which was responsible for configuring as input for short time). Did that already solve your problem? BTW: Software I2C (plus small hardware configuration-doc :-) ) sounds interesting!

malosek, 2016-04-27:
Hello,
sure solution is easy, was just "investigating" if is acceptable:) I'll prepare PR with this one and soft i2c as well.
wbr
malo

PeterKietzmann, 2016-06-02:
@malosek can we close this issue? The i2c PR is already there. About the GPIO stuff we found a solution as well, didn't we?

PeterKietzmann, 2016-06-28:
(Solution was already found. memo label set not to forget the proposal to fix problem on multiple `gpio_init()` calls )