RIOT-OS/RIOT #5304: stm32: Uart driver inconsistent

toonst, 2016-04-13:
If you look at the uart driver implementation on the stm32 cpu's, there are two ways this is implemented.
- For F0, F3 and L1:
Similar to other peripherals, some macro's are defined in `<board>/include/periph_conf.h`:
```
/* UART 0 device configuration */
#define UART_0_DEV          USART2
#define UART_0_CLKEN()      (RCC->APB1ENR |= RCC_APB1ENR_USART2EN)
#define UART_0_CLKDIS()     (RCC->APB1ENR &= (~RCC_APB1ENR_USART2EN))
#define UART_0_IRQ          USART2_IRQn
#define UART_0_ISR          isr_usart2
/* UART 0 pin configuration */
#define UART_0_PORT         GPIOA
#define UART_0_PORT_CLKEN() (RCC->AHBENR |= RCC_AHBENR_GPIOAEN)
#define UART_0_RX_PIN       3
#define UART_0_TX_PIN       2
#define UART_0_AF           1
```
- For F1 and F4 (and F2 [here](https://github.com/RIOT-OS/RIOT/pull/4497)):
`static const uart_conf_t uart_config[]` is defined in `<board>/include/periph_conf.h`:
```
static const uart_conf_t uart_config[] = {
    {
        .dev        = USART3,
        .rcc_mask   = RCC_APB1ENR_USART3EN,
        .rx_pin     = GPIO_PIN(PORT_D, 9),
        .tx_pin     = GPIO_PIN(PORT_D, 8),
        .af         = GPIO_AF7,
        .irqn       = USART3_IRQn,
        .dma_stream = 3,
        .dma_chan   = 4
    }
};

/* assign ISR vector names */
#define UART_0_ISR          isr_usart3
#define UART_0_DMA_ISR      isr_dma1_stream3

/* deduct number of defined UART interfaces */
#define UART_NUMOF          (sizeof(uart_config) / sizeof(uart_config[0]))
```
The driver gets the the correct configuration by using this array, so you don't need any `#if UART_0_EN` nor switch case in the driver implementation (`cpu/stm32f1/periph/uart.c`).

Is there a good reason these are implemented differently? If no, what is the preferred way to implement it?

kaspar030, 2016-04-13:
The array-based one is the preferred way. Before, a lot of defines have been used. Not all drivers have been updated...

lebrush, 2016-04-13:
The first case is the old style configuration. Some drives were updated to a new style format but not all. Feel free to update the drivers of F0, F3 and L1 :-)

kaspar030, 2016-04-13:
> Feel free to update the drivers of F0, F3 and L1 :-)

Would be awesome if they're similar enough to make them share (most of
the) code. ;)

toonst, 2016-04-14:
Ok, thanks for the response!

closed 2016-04-14 by toonst