RIOT-OS/RIOT #5312: samd21/PA27 GPIO not linked to extint 15

bapclenet, 2016-04-13:
According to the datasheet PA27 has for interrupt 15 but samd21/gpio.c:
```
static const int8_t exti_config[2][32] = {
    {-1,  1, -1, -1,  4,  5,  6,  7, -1,  9, 10, 11, 12, 13, 14, 15,
     -1,  1,  2,  3, -1, -1,  6,  7, 12, 13, -1, -1,  8, -1, -1, -1},
    { 0, -1,  2,  3, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1,
      0,  1, -1, -1, -1, -1,  6,  7, -1, -1, -1, 15,  8, -1, 10, 11},
};
```

This should be:
````
static const int8_t exti_config[2][32] = {
    {-1,  1, -1, -1,  4,  5,  6,  7, -1,  9, 10, 11, 12, 13, 14, 15,
     -1,  1,  2,  3, -1, -1,  6,  7, 12, 13, -1, 15,  8, -1, -1, -1},
    { 0, -1,  2,  3, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1,
      0,  1, -1, -1, -1, -1,  6,  7, -1, -1, -1, 15,  8, -1, 10, 11},
};
```

haukepetersen, 2016-04-14:
Ups, overlooked that you provided a patch already. Also PA30 and PA31 are not mapped correctly, all three of them are wrongly switched with PB27, PB30, and PB31.

haukepetersen, 2016-04-20:
fixed with #5315

closed 2016-04-20 by haukepetersen