RIOT-OS/RIOT #5318: netstats: add stats to more modules

OlegHahm, 2016-04-14:
Currently only L2 netstats are available and only for some drivers.
The following modules should provide netstats, too:
 - [] GNRC IPv6
 - [] GNRC UDP
 - [] ethos
 - [] xbee
 - [] GNRC SLIP

Optionally also:
 - [] lwip
 - [] emb6

kaspar030, 2016-04-14:
> # [] GNRC IPv6
> # [] GNRC UDP

Nice, soon we'll surpass Linux in what we know about what's going on in
the network stack!

OlegHahm, 2016-04-14:
Sounds like a reasonable goal for an OS that is developed by a lot of people with a background in research on computer networks, doesn't it? ;-)

kYc0o, 2016-07-26:
Updated and moved for the next release.

bergzand, 2017-01-10:
I was looking at a way to implement the GNRC UDP stats since it doesn't seem te be a lot of work and it looks like an easy way to learn something about the network stack.

First, there doesn't seem to be some kind of struct in the UDP handler to store the stats in. Am I right in this or have I just missed it?

Second: the IPv6 stats does not seem to care about errors in the lower layers, such that for example, if a ping reply is assumed lost according to the transceiver, the L2 stats can count an error in the TX stats, but the IPv6 stats don't show a TX error.
Is this by design and wanted? Or is an implementation where the upper layer also counts an error wanted?

OlegHahm, 2017-01-10:
> First, there doesn't seem to be some kind of struct in the UDP handler to store the stats in. Am I right in this or have I just missed it?

In principle `sock` could hold this information, I guess.

> Second: the IPv6 stats does not seem to care about errors in the lower layers, such that for example, if a ping reply is assumed lost according to the transceiver, the L2 stats can count an error in the TX stats, but the IPv6 stats don't show a TX error.
Is this by design and wanted? Or is an implementation where the upper layer also counts an error wanted?

My design idea was that the layer statistics count only what is under their control. I.e. if IPv6 passes a datagram to the link layer, it assumes the packet to be sent -> TX success. If it cannot pass the datagram to the lower layer, because, for instance, no proper interface could be determined, it would count it as a TX error.

miri64, 2017-01-10:
>> First, there doesn't seem to be some kind of struct in the UDP handler to store the stats in. Am I right in this or have I just missed it?
>
> In principle `sock` could hold this information, I guess.

Please don't mix-in `sock` into `gnrc_udp`. The latter should be able to be used without `sock_udp`. All things considered the `gnrc_udp` can be considered as singleton (having multiple `gnrc_udp` instances doesn't make much sense), so you could just store this kind of data statically in the module I would propose.

bergzand, 2017-01-10:
> My design idea was that the layer statistics count only what is under their control. I.e. if IPv6 passes a datagram to the link layer, it assumes the packet to be sent -> TX success. If it cannot [...]

Makes sense, thanks for the explanation.

> Please don't mix-in sock into gnrc_udp. The latter should be able to be used without sock_udp. All things considered the gnrc_udp can be considered as singleton (having multiple gnrc_udp instances [...]

Sounds doable. I'll look into this in the next few days when I have time.
Thanks for all the explanations

OlegHahm, 2017-01-10:
> > > First, there doesn't seem to be some kind of struct in the UDP handler to store the stats in. Am I right in this or have I just missed it?

> >    In principle sock could hold this information, I guess.

> Please don't mix-in sock into gnrc_udp. The latter should be able to be used without sock_udp. All things considered the gnrc_udp can be considered as singleton (having multiple gnrc_udp instances [...]

The question was about UDP, not about gnrc_udp. That's why I proposed sock. Also, I thought that statistics per peer might be more interesting than general UDP statistics.

miri64, 2017-01-11:
> The question was about UDP, not about gnrc_udp. That's why I proposed sock. Also, I thought that statistics per peer might be more interesting than general UDP statistics.

But @bergzand was asking about *GNRC* UDP, so your comment might have been misunderstood (as I did). Regarding the per-peer statistics it might indeed be more useful (and easier to implement) to store the data in GNRC's `sock_udp` struct and provide a (GNRC-specific!) function to access that.

bergzand, 2017-01-11:
So there are two options here? Am I understanding them correct?
One as static variables in the gnrc_udp  code with the result that only total UDP packets/bytes are counted.
Second option: add the statistics to the sock_udp and get per listening/connecting socket stats

I don't really have a preference for one or the other. With the second option I don't know yet how the gnrc code or the shell commands would query the data, but that's probably not the hardest part to figure out.
Also, with the second option, if the socket is closed the stats would be lost right? Is there a problem with implementing both to solve this? So get a generic udp stats struct for the total number of udp packet/bytes and per connection counters?

OlegHahm, 2017-01-11:
So, maybe we should discuss what we actually want to achieve, before we discuss implementation details. We currently have IPv6 statistics which are stored per interface. Since UDP doesn't provide any additional services in terms of reliability, we could get three additional insights from gathering UDP stats as well:
1. How much of the IPv6 traffic (summing up over all interfaces) was actually UDP traffic.
2. A UDP datagram has been tried to be sent but failed on the IP layer (e.g., if no matching interface could be found).
3. How much traffic per communication peer we have.

I think all three insights can be achieved by storing the statistics inside the sock struct, while only 1. and 2. can be achieved if we store UDP stats centrally. Furthermore, I would think that storing these stats in the sock struct itself (if something like this exists) rather than in a GNRC specific struct. The implementation of counting the datagrams probably has to happen inside GNRC (or whatever network stack you prefer).

maribu, 2022-09-16:
netstats are now collected at the netif level, so drivers do not need to be extended.

emb6 has been dropped, so no need to extend it. lwip does not yet have netstats, but given how easy it is to add them, there seems to be no demand.

If anyone disagrees on closing this one, please reopen.

closed 2022-09-16 by maribu