RIOT-OS/RIOT #5325: Threads unstable behavior

srisler, 2016-04-15:
Hi all,

I am developing on a STM32F3Discovery board, connecting to a NRF24L01 and an ESP8266. It is used as a gateway that create a link between internet (through Wifi) and a local wireless sensor network. 

The system is composed of 6 threads (including main and idle), and basically takes a message received from a side and send it to the other side (and the other way around).

It works pretty well, but sometimes it suddenly stops: it stops working, LEDs stop blinking... but I got response to PS and other shell commands, then I conclude, the system is not "crashed" but threads are just stops. "ps" response is totally normal (thread status, memory used...)

I observe this behavior when the system is running for 1hour average.. (sometimes 30min sometimes 2h...)

Have you ever experience this? Do you have any advice or point to check? 

Thank you in advance,

OlegHahm, 2016-04-15:
If the shell is still working, then probably one of the threads is somehow deadlocked. What is ps telling you about the current state of the threads?

P.S. Side question, do you plan to PR the driver for the ESP8266?

srisler, 2016-04-15:
Thank you for quick reply. I wait for the system to be in a "blocked" state, and I past the ps response.

About the ESP driver, it would be a pleasure to share it. Honestly, I am a newbie in open source sharing, and I hope my code is not too much "home made". Well, first of all, I would like to be sure that it is not the cause of my problem ;)

srisler, 2016-04-15:
Here it is:
```
2016-04-15 21:53:10,760 - INFO # ps
2016-04-15 21:53:10,769 - INFO # 	pid | name                 | state    Q | pri | stack ( used) | location   
2016-04-15 21:53:10,775 - INFO # 	  1 | idle                 | pending  Q |  15 |   256 (  136) | 0x20000204 
2016-04-15 21:53:10,788 - INFO # 	  2 | main                 | running  Q |   7 |  1536 (  660) | 0x20000304 
2016-04-15 21:53:10,790 - INFO # 	  3 | WD manager           | bl mutex _ |   4 |   728 (  372) | 0x20002294 
2016-04-15 21:53:10,815 - INFO # 	  4 | Wifi manager         | bl mutex _ |   8 |   768 (  256) | 0x200027dc 
2016-04-15 21:53:10,818 - INFO # 	  5 | TCP manager          | bl mutex _ |   4 |  2048 (  564) | 0x200015ec 
2016-04-15 21:53:10,819 - INFO # 	  6 | Com manager          | bl mutex _ |   6 |  1024 (  604) | 0x20003670 
2016-04-15 21:53:10,820 - INFO # 	  7 | Asap_Prot            | bl rx    _ |   1 |  1536 (  328) | 0x20003adc 
2016-04-15 21:53:10,821 - INFO # 	    | SUM                  |            |     |  7896 ( 2920)
```

OlegHahm, 2016-04-15:
Yes, looks indeed like a classical deadlock. All threads (except main and idle are blocked either because they're waiting for a mutex or for an incoming IPC message). Without further knowledge of your code is difficult to make any more precise analysis. The thread "Asap_Prot" is running the event loop to wait for received packets from the driver?

srisler, 2016-04-15:
Yes. Asap_prot is always waiting for a message coming from Nrf24l01. The ps response is exactly the same during normal execution.  

The 4 other one have the same behavior. In a while(1):
* msg_try_receive for an incoming IPC message
* Check for the time to let blink their assigned Led. (1s period)
* 500µs xtimer_usleep between each iteration

ESP8266 incoming message are handled in TCP manager. However, the instance of esp8266 is allocated in the main and the reference is passed to Wifi Manager and TCP manager that use this shared resource...

What can I check to track down the problem?

OlegHahm, 2016-04-15:
Does `make debug` work on this board? (I'm not so familiar with the STM32 platforms.)

srisler, 2016-04-15:
I have never tried actually (trace of shell were enough so far). I'll give it a try asap

srisler, 2016-04-15:
Here the error I have after installing OpenOCD:
```
command -v /home/seb/Documents/WW/RIOT/dist/tools/openocd/openocd.sh >/dev/null 2>&1 || \
		{ /bin/echo -e \
		'\033[1;31m Debug program /home/seb/Documents/WW/RIOT/dist/tools/openocd/openocd.sh not found. Aborting.\033[0m'; \
		exit 1; }
/home/seb/Documents/WW/RIOT/dist/tools/openocd/openocd.sh debug
### Starting Debugging ###
/home/seb/Documents/WW/RIOT/dist/tools/openocd/openocd.sh: 214: /home/seb/Documents/WW/RIOT/dist/tools/openocd/openocd.sh: arm-none-eabi-gdb: not found
/home/seb/Documents/WW/RIOT/dist/tools/openocd/openocd.sh: 218: kill: Usage: kill [-s sigspec | -signum | -sigspec] [pid | job]... or
kill -l [exitstatus]
Open On-Chip Debugger 0.10.0-dev-00251-g888d5a5 (2016-04-15-22:50)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
```
However, arm-none-eabi-gdb --version works well

OlegHahm, 2016-04-15:
Strange.

srisler, 2016-04-30:
Ok. I finally managed to make it work. When the system looks locked, program is spinning in:
```
    while (1) {
        if (lpm_prevent_sleep) {
            lpm_set(LPM_IDLE);
        }
```
In kernel_init.c

Which is the Idle thread, then to me it is .... normal.

I'll investigate a bit more to get some more data.

Thanks for help,

srisler, 2016-05-02:
Hi guys,

I finally managed to make the Debug work (thanks to code of #5175). Program is spinning in Idle thread while(1): idle_thread() called by sched_switch()

All other thread are not running anymore.... any clue?

Thks

authmillenon, 2016-05-02:
Do your threads run in an endless loop? Otherwise, they just terminate when their task is done => it goes into idle mode (which currently is the spinning you observered on the platform you use due to missing LPM support).

srisler, 2016-05-02:
Yes, all my thread run in a while(1). Is there other reason they would terminate? Would they appear in PS if they are terminated ? (because they are)

srisler, 2016-05-05:
Hi,

Up ?

Any lead to investigate ?

Thx for any help

kaspar030, 2016-05-05:
Could you add some debug output to xtimer_usleep(), like at the top and at the bottom? The goal is to find out whether you're hitting a problem with xtimer.

kaspar030, 2016-05-05:
> I observe this behavior when the system is running for 1hour average..

Ah, printing sth. every 500us for one hour is maybe not a good idea. I'll think of sth.

srisler, 2016-05-05:
I am using xtimer_now a lot (blinking led, connection retry...). 

I have a xtimer_usleep(500) in each iteration of the while(1) loops. Is it good practise?

Thx

malosek, 2016-05-05:
Hello,
could you try to run your code with changes from #5355 ?
wbr
malo

srisler, 2016-05-05:
I tried. It results a kernel panic (hardfault, stack overflowed by 8 bytes). 

To be clear, I just took the change of the 3 files of the commit #5355.

malosek, 2016-05-05:
Always crashing at the same place/thread? can you post the backtrace? did you try to increase stack for that particular thread?
wbr
malo

srisler, 2016-05-05:
I have not the backtrace. I try to make another run with and I'll update.

I just notice that if I wait long enough (more than an hour... but seems pretty random) all the thread resume... Then they are definitely not terminated, that's why they appear in ps....

Is there something to check in debug mode in the scheduler to know a bit more about the state of these threads?

Thx

OlegHahm, 2016-05-05:
You can either try with https://github.com/RIOT-OS/RIOT/pull/4058 or use `call ps()` from gdb.

srisler, 2016-05-05:
It looks great and will help a lot. I can't test it till next week, but I will do as soon as I can.

I'll let you know,

kaspar030, 2016-05-05:
> I just notice that if I wait long enough (more than an hour... but seems
> pretty random) all the thread resume..

It's not like exactly 71 minutes?

srisler, 2016-05-06:
uhm... Actually yes and it is pretty accurate. What does it mean?

kaspar030, 2016-05-06:
~71 minutes are 2**32 microseconds. That means somehow the xtimer overflow is not handled correctly.

srisler, 2016-05-06:
Ok. It explains why the system is "locked" for 71 minutes and then start
again. But not the reason why it gets lock. Am I right?

kaspar030, 2016-05-06:
> Ok. It explains why the system is "locked" for 71 minutes and then start
> again. But not the reason why it gets lock. Am I right?

Exactly... I'll look into it.

srisler, 2016-05-06:
Ok thx. To be clear, here how I deal with xtimer:

xtimer_init() is called in main.

Each thread has this structure:
```
while(1){
//Several calls to xtimer_now() to manage timing

xtimer_usleep(500);
}
```

malosek, 2016-05-07:
Hello,
could you add some debug printf/puts in the condition _xtimer_set_absolute
if (target < now) {
        timer->long_target++;
}
and let it run? if you get it printed after ~71 mins...
wbr
malo

srisler, 2016-05-11:
Hi,

Sorry, I was away till now. I added a "puts" in the condition. 

It is reached quite often actually. Most of the time it "locks" only one thread. FYI, I coded a "watchdog system" which is an independent thread that must receive messages from every other threads, otherwise it reboots the all system. When the condition is reached  (target < now), it locks only one thread, then the watchdog reboot the system. It is so quick that I haven't seen that before. 

I guess that all threads are locked (even watchdog) when the system is locked for 71 min (I am waiting for it to happen).

PS: I am wondering if the "puts" has side effect, because it is reached quite often and ALWAYS makes the system reboot.

malosek, 2016-05-11:
Hello,

The condition should be true if you are setting timer target with 32 bits wrap. like setting lower 32bits of absolute timer to 0xFF while now is 0xFFFF so it needs to add one to higher 32bits long_target to make 64 bits absolute target time.
Im surprised that it happens "quite often" since with 1us timer you have 32bits wrap in 71 minutes.
Does it happen much sooner?

You could add some asserts there since only after 71 mins _long_cnt should change to 1, just check  it in 
timer->long_target = _long_cnt;
if (target < now) {
        timer->long_target++;
}
from your description seems like  timer->long_target++; is not correct (2 instead of 1), just try to add assert checking if timer->long_target is 2 after the ++ operation. it should assert in 71 mins in your case.
(if my theory is correct of course:)

wbr
malo

srisler, 2016-05-11:
I added the assertion. I am waiting for it to happen.

> Does it happen much sooner?
Yes, it can happen several minutes/seconds after it starts

srisler, 2016-05-11:
++ seems to work fine:
```
2016-05-11 14:26:35,004 - INFO # xtimer_set_absolute(): Condition (target < now) reached !! Long target is : 0
2016-05-11 14:26:35,008 - INFO # xtimer_set_absolute(): long_target is now : 1
```

Code is 
```
    if (target < now) {
    	printf("xtimer_set_absolute(): Condition (target < now) reached !! Long target is : %ld\n", timer->long_target);
        timer->long_target++;
        printf("xtimer_set_absolute(): long_target is now : %ld\n", timer->long_target);
    }
```

malosek, 2016-05-11:
Hello,
and it still hangs for 71 mins after first 71 mins?
just to be sure print even timer->target and now there.
wbr
malo

srisler, 2016-05-11:
Actually, there are several behaviours: 

Case 1 : the system works fine for 1 to 30 minutes (roughly). Then it gets block for 71 minutes. Then it gets back to normal for 1 to 30 minutes, etc... (no reboot)
 
Case 2 : One thread gets locked but not the other. Apparently the cause is from xtimer (target < now). Watchdog reboots the system.

Case 3 (rare): Exactly the same that case 1, but the system never resume.

I am surprised about this unstable state. Is xtimer correctly used ? Is it specific to the board?

> xtimer_init() is called in main.
Each thread has this structure:
while(1){
//Several calls to xtimer_now() to manage timing
xtimer_usleep(500);
}

malosek, 2016-05-11:
Hello,
you are using it correctly.
seems like you are not hitting the issue which I was thinking of, then just check general things like stack sizes, printf formatting strings and dive deeper:)
btw: there is #5428 PR which should be big step forward with xtimer.
wbr
malo

srisler, 2016-05-11:
Stack sizes are largely oversized, in order to not be bother with . I check and re-check mutex. This bug is so random, it is difficult to dive deeper (at least to know where to dive). And apparently there is something (maybe just a part) with xtimer. I'll wait for the PR and I'll see. 

Btw, case 2 just happens:
```
2016-05-11 15:44:58,136 - INFO # xtimer_set_absolute(): Condition (target < now) reached !! 
2016-05-11 15:44:58,137 - INFO # Long target is : 0
2016-05-11 15:44:58,139 - INFO # Target:881329256
2016-05-11 15:44:58,141 - INFO # Now:881331551
2016-05-11 15:44:58,142 - INFO # xtimer_set_absolute(): 
2016-05-11 15:44:58,144 - INFO # Long target is : 1
2016-05-11 15:44:58,146 - INFO # Target:881329256
2016-05-11 15:44:58,147 - INFO # Now:881331551
```

srisler, 2016-05-11:
And again:
```
2016-05-11 15:55:25,140 - INFO # xtimer_set_absolute(): Condition (target < now) reached !! 
2016-05-11 15:55:25,142 - INFO # Long target is : 0
2016-05-11 15:55:25,143 - INFO # Target:270050128
2016-05-11 15:55:25,144 - INFO # Now:270051856
2016-05-11 15:55:25,146 - INFO # xtimer_set_absolute(): 
2016-05-11 15:55:25,148 - INFO # Long target is : 1
2016-05-11 15:55:25,149 - INFO # Target:270050128
2016-05-11 15:55:25,150 - INFO # Now:270051856
```

malosek, 2016-05-11:
Hello,
your system is too busy, you are getting too late to into the _xtimer_set_absolute function.
increase xtimer_usleeps like 10 times and should work, you can try to play with XTIMER_BACKOFF as well.
wbr
malo

srisler, 2016-05-11:
Ok, I increase to 5ms. I'll let you know

kaspar030, 2016-05-11:
Could you try merging #5428?

srisler, 2016-05-12:
Hi.

System is running almost perfectly since yesterday. Then, @malosek you were right, the system was too busy... 

However, I observe one "pause" of 71 minutes since yesterday. @kaspar030, I am away till next tuesday. I'll try to merge #5428 then. I'll let the system running.

Thank you for your help and time

closed 2016-05-12 by srisler