RIOT-OS/RIOT #5328: GNRC: remove neighbor cache from minimal configuration

OlegHahm, 2016-04-18:
If I remember correctly, it was not possible to build the gnrc_minimal example without neighbor cache because even for link-local communication over 6LoWPAN NDP is required, while it should work also without.

authmillenon, 2016-04-18:
That's because the reverse-translation is implemented in [6Lo-ND](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/sixlowpan/nd/gnrc_sixlowpan_nd.c#L196-L211). Maybe a cleaner solution that would be helpful with this solution would be to provide this in a separate module, that anyone (including 6Lo-ND, RPL, and `gnrc_minimal`) can use.

OlegHahm, 2016-04-18:
Yes, I know, I just realized that there has been no issue for this yet.

kYc0o, 2016-07-26:
Maybe it can be addressed on the new NDP design? For the next release if possible :)

miri64, 2019-08-10:
This should have been fixed via the NIB rework. Reopen if you disagree.

closed 2019-08-10 by miri64