RIOT-OS/RIOT #5371: Support for IP packet fragmentation/reassembly

oberstet, 2016-04-20:
As from a discussion on IRC, I learned that RIOT currently does not support IP packet fragmentation/reassembly.

RFC4944 though seems to [suggest](https://tools.ietf.org/html/rfc4944#section-5.3) not to have IP fragmentation/reassembly independent of the link layer fragmentation of 6LoWPAN, but to "collapse" that:

> If an entire payload (e.g., IPv6) datagram fits within a single
   802.15.4 frame, it is unfragmented and the LoWPAN encapsulation
   should not contain a fragmentation header.  If the datagram does not
   fit within a single IEEE 802.15.4 frame, it SHALL be broken into link
   fragments.

As it stands, the lack of that in RIOT makes it impossible to use a full UDP datagram payload (64k), but limits the latter to <1280 (the IPv6 required minimum MTU).

authmillenon, 2016-04-20:
Should be `<= 1280` (did not check), since 1280 is the [defined MTU for 6LoWPAN](https://tools.ietf.org/html/rfc4944#section-4.

authmillenon, 2016-04-20:
But regardless, yes: IPv6 fragmentation should be implemented at one point.

oberstet, 2016-04-20:
@authmillenon "IPv6 fragmentation should be implemented" cool!

rgd "<" vs "<=": what's the _theoretical_ usable UDP datagram payload left when there is no support for IP frag/reas? I thought that necessarily would be strictly less than 1280 because of headers?

Also (from IRC):

> <eintopf> Most people says maximum 4-5 fragments, nothing more for 802.15.4 6lowpan networks

So that would make up even less _practically_ usable UDP datagram payload. Something like 500-1000 octets ..

And does above IRC comment suggest at that IP frag/reas. would be _practically_ useless anyway?

Because:

> <eintopf> I transmitted one day 64k over 2.4 Ghz 802.15.4 6lowpan
 <eintopf> took 10 seconds
 <eintopf> rtt

kYc0o, 2016-07-26:
Maybe this could be addressed on the next release?

miri64, 2018-10-24:
Since @Lotterleben wants to do more with RIOT again, I asked her if she wants to handle this issue, and she said yes.

J4ken, 2019-05-28:
Hi! Is there any updates on this issue?

miri64, 2019-05-28:
I'm currently working on it. #11593 is a starting point for this and I plan to provide initial support for reassembly today.

miri64, 2019-11-29:
#11596 and #11623 were merged a while back, so this is fixed.

closed 2019-11-29 by miri64