RIOT-OS/RIOT #5382: Nucleo F446RE

wilhelmmatilainen, 2016-04-21:
Could Nucleo STM32F446RE (http://www.st.com/web/en/catalog/mmc/SC1169/SS1577/LN11875/PF260537) be added to supported boards?

I would do this myself, if I knew what exactly should be done.
Maybe I could get it running for myself, but creating then a pull request for something that "seems" to work probably isn't the right way.

I just bought this board and would like to use Riot OS on it.

toonst, 2016-04-25:
Porting to a new board is pretty straight forward. Have a look here:
https://github.com/RIOT-OS/RIOT/wiki/Porting-Guide

There's already support for nucleo-f401, so you can start from that.

DipSwitch, 2016-04-26:
Sorry for the delay! This is een Nucleo 64 or 144? If its a 64 the board
port is identical to the nucleo-f4 if it's an 144 it would need some
tweaking if i'm correct, but not that much. I'll try to find some time this
week to have a look.

Cheers!

wilhelmmatilainen, 2016-04-26:
It is 64.

So, all the different parameters are identical?

DipSwitch, 2016-04-30:
I think it would suffice to copy. You could try this branch: https://github.com/DipSwitch/RIOT/tree/pr/add_nucleo_f446

aabadie, 2016-07-29:
I was not aware that this issue was already opened. I initiate a port to nucleo f446 in #5706, feel free to have a look.

miri64, 2016-10-18:
#5706 was merged, so close?

closed 2016-10-18 by miri64