RIOT-OS/RIOT #5389: Build on master failed

smlng, 2016-04-22:
Hi all,

I currently have an issue compiling and running my test application (see [here](https://github.com/smartuni/GirlsDay2016/tree/master/src/sensor_node)) using latest RIOT master (77cf00bd4bd): this results in a kernel panic.

```
2016-04-22 14:38:00,939 - INFO # ipv6: waiting for incoming message.
2016-04-22 14:38:00,944 - INFO # 6lo nd: retransmit multicast rtr sol in 10 sec
2016-04-22 14:38:00,950 - INFO # ndp internal: send router solicitation (iface: 6, dst: ff02::2)
2016-04-22 14:38:00,957 - INFO # ndp: building source link-layer address option (l2addr: 59:b3:4e:48:6f:dc:60:02)
2016-04-22 14:38:00,960 - INFO # ndp: building router solicitation message
2016-04-22 14:38:00,965 - INFO # icmpv6: Building ICMPv6 message with type=133, code=0
2016-04-22 14:38:00,968 - INFO # ipv6: GNRC_NETAPI_MSG_TYPE_SND received
2016-04-22 14:38:00,973 - INFO # ipv6: set payload length to 24 (network byteorder 1800)
2016-04-22 14:38:00,976 - INFO # ipv6: set next header to 58
2016-04-22 14:38:00,980 - INFO # ipv6: calculate checksum for upper header.
2016-04-22 14:38:00,982 - INFO # ipv6: send multicast over interface 6
2016-04-22 14:38:00,985 - INFO # ipv6: send to 6LoWPAN instead
2016-04-22 14:38:00,993 - INFO # ipv6: waiting for incoming message.
2016-04-22 14:38:00,999 - INFO # main(): This is RIOT! (Version: 2016.07-devel-89-g77cf0-Sebastians-Air.lan)
2016-04-22 14:38:01,002 - INFO # Girls Day 2016 @ HAW Hamburg!
2016-04-22 14:38:01,003 - INFO # ================
2016-04-22 14:38:01,007 - INFO # You are running RIOT on a(n) pba-d-01-kw2x board.
2016-04-22 14:38:01,011 - INFO # This board features a(n) kw2x MCU.
2016-04-22 14:38:01,013 - INFO # ================
2016-04-22 14:38:01,013 - INFO # . init network
2016-04-22 14:38:01,013 - INFO # .
2016-04-22 14:38:01,016 - INFO # .. init sensors
2016-04-22 14:38:01,132 - INFO # :
2016-04-22 14:38:01,133 - INFO # ... init coap
2016-04-22 14:38:01,134 - INFO # .
2016-04-22 14:38:01,134 - INFO # :
2016-04-22 14:38:01,136 - INFO # .... init shell
2016-04-22 14:38:01,136 - INFO # :
2016-04-22 14:38:01,136 - INFO # :
2016-04-22 14:38:01,137 - INFO # > 0x2b2b
2016-04-22 14:38:01,139 - INFO # *** RIOT kernel panic:
2016-04-22 14:38:01,140 - INFO # FAILED ASSERTION.
2016-04-22 14:38:01,141 - INFO # 
2016-04-22 14:38:01,147 - INFO # 	pid | name                 | state    Q | pri | stack ( used) | location   
2016-04-22 14:38:01,153 - INFO # 	  1 | idle                 | pending  Q |  15 |   256 (  128) | 0x1fffc2fc 
2016-04-22 14:38:01,160 - INFO # 	  2 | main                 | bl mutex _ |   7 |  1536 (  484) | 0x1fffc3fc 
2016-04-22 14:38:01,167 - INFO # 	  3 | 6lo                  | bl rx    _ |   3 |  1024 (  400) | 0x200000a4 
2016-04-22 14:38:01,173 - INFO # 	  4 | ipv6                 | bl rx    _ |   4 |  1536 (  536) | 0x1fffe018 
2016-04-22 14:38:01,180 - INFO # 	  5 | udp                  | bl rx    _ |   5 |  1024 (  264) | 0x200008ac 
2016-04-22 14:38:01,187 - INFO # 	  6 | kw2xrf               | bl rx    _ |   3 |  1024 (  240) | 0x1fffdc14 
2016-04-22 14:38:01,194 - INFO # 	  7 | sensor_thread        | running  Q |   7 |  1024 (  436) | 0x1fffcd14 
2016-04-22 14:38:01,200 - INFO # 	  8 | coap_thread          | pending  Q |   7 |  1024 (  124) | 0x1fffd730 
2016-04-22 14:38:01,208 - INFO # 	    | SUM                  |            |     |  8448 ( 2612)
2016-04-22 14:38:01,208 - INFO # 
2016-04-22 14:38:01,209 - INFO # *** halted.
2016-04-22 14:38:01,209 - INFO # 
```

However, using the release branch `2016.04-branch` (1c96496e3689) everything is fine, my app compiles and runs.

authmillenon, 2016-04-22:
This is a failed assertion. Can you remove `core/include/assert.h` and dump the output if you compile and run it again?

cgundogan, 2016-04-22:
@smlng can you check `BOARD=x make objdump` and check the address `0x2b2b`?

smlng, 2016-04-22:
FYI: the `gnrc_networking` is running with master and release-branch, no kernel panic there. The failure seems to be with my app only - I use some sensor drivers and ADC stuff ...

authmillenon, 2016-04-22:
Please check with one of our proposed scenarios.

smlng, 2016-04-22:
Removing `core/include/assert.h` does help, no kernel panic anymore. My tool is running on master branch as well.

cgundogan, 2016-04-22:
>Removing core/include/assert.h does help, no kernel panic anymore.

It does not help (: The intention was actually to find out which assertion failed.

vincent-d, 2016-04-26:
I've got a failed assertion as well recently and it came from the new assert in the msg_queue_init added in bb35913. Maybe it's the same here for you.

smlng, 2016-04-27:
ah thanks @vincent-d, I'll look into it - didn't have time to check that, yet.

OlegHahm, 2016-05-31:
What's the latest?

smlng, 2016-06-02:
Well the issues was prevalent with my specific code and RIOT master branch. However, following the hint by @vincent-d I found that the failing assert is in [cib.h](https://github.com/RIOT-OS/RIOT/blob/master/core/include/cib.h#L55), the message queue size has to be a power of 2. However I used a size of 10 (chosen by careful dice role 😄 ), changing this to 8 resolves the issue. 

I don't know the reasoning behind this power-of-2 size for the message queue, but a kernel panic is kind of an exaggeration - maybe a compiler warning/error would be more appropriate then panic on runtime. Is there any documentation on this?

TL;DR: size of message queue has to be power of 2. We may close this issue?!

kaspar030, 2016-06-02:
> I don't know the reasoning behind this power-of-2 size for the message queue,

The reason is that cib uses eficient bit masks instead of modulo operations, which only works for power-of-two. IMHO that constraint is documented both for cib and message queues. Not sure it is compile-time assertable.

cgundogan, 2016-06-02:
PEBCAK

closed 2016-06-02 by cgundogan

OlegHahm, 2016-06-02:
> Is there any documentation on this?

http://doc.riot-os.org/group__core__msg.html#ga480e6f32c8ab18579b62a890f3fda2cd

Do you want it red and blinking? :stuck_out_tongue_winking_eye: 
But yes, checking at compile time should be feasible.

smlng, 2016-06-02:
Ah darn, its even in bold ... I really should find some spare time to read more 📖