RIOT-OS/RIOT #5390: pkg: OpenWSN does not compile

authmillenon, 2016-04-22:
There are references to very old modules (like hwtimer.h) in `openwsn` so it does not compile. Test setup: compile https://github.com/RIOT-OS/applications/tree/master/openwsn.

kYc0o, 2016-07-11:
Do you refer only that the package does not compile or that the application openwsn does not compile? if that's the case, it is not an issue that would be better to discuss in the applications repo? It actually exists an [issue](https://github.com/RIOT-OS/applications/issues/27) pointing to the same.

kYc0o, 2016-07-26:
The issue about old modules is not present anymore, however the package is still not building. Postponing and I'll try to open another issue pointing to the new problem, afterwards we can close this one.

miri64, 2016-07-26:
> The issue about old modules is not present anymore, however the package is still not building.

Here at least it's still the hwtimer problem.

OlegHahm, 2016-07-26:
Depends on how you want to handle this, but I always preferred to keep unsolved issues tagged for the release in order to easily gather them for the "known issues" list.

PeterKietzmann, 2017-01-31:
OpenWSN ahs been removed #6359

closed 2017-01-31 by PeterKietzmann