RIOT-OS/RIOT #5394: lwIP: not able to receive

authmillenon, 2016-04-25:
For some reason a node with `tests/lwip` flashed isn't able to receive any packets on `samr21-xpro` (the interrupt for reception on driver level isn't even fired). With bisecting I was able to pin-point it to https://github.com/RIOT-OS/RIOT/pull/5261. With the help of @OlegHahm I was able to see that the `pending_tx` counter never actually reached 0, but I don't know why.

Here's patch for checking `pending_tx`'s value:

```diff
diff --git a/drivers/at86rf2xx/at86rf2xx.c b/drivers/at86rf2xx/at86rf2xx.c
index 1dc0bec..fb19c36 100644
--- a/drivers/at86rf2xx/at86rf2xx.c
+++ b/drivers/at86rf2xx/at86rf2xx.c
@@ -44,6 +44,7 @@ void at86rf2xx_setup(at86rf2xx_t *dev, const at86rf2xx_params_t *params)
     memcpy(&dev->params, params, sizeof(at86rf2xx_params_t));
     dev->idle_state = AT86RF2XX_STATE_TRX_OFF;
     dev->state = AT86RF2XX_STATE_SLEEP;
+    printf("%s():%d Initializing pending_tx with 0\n", __func__, __LINE__);
     dev->pending_tx = 0;
     /* initialise SPI */
     spi_init_master(dev->params.spi, SPI_CONF_FIRST_RISING, params->spi_speed);
@@ -105,9 +106,7 @@ void at86rf2xx_reset(at86rf2xx_t *dev)
     at86rf2xx_set_option(dev, AT86RF2XX_OPT_CSMA, true);
     at86rf2xx_set_option(dev, AT86RF2XX_OPT_TELL_RX_START, false);
     at86rf2xx_set_option(dev, AT86RF2XX_OPT_TELL_RX_END, true);
-#ifdef MODULE_NETSTATS_L2
     at86rf2xx_set_option(dev, AT86RF2XX_OPT_TELL_TX_END, true);
-#endif
     /* set default protocol */
 #ifdef MODULE_GNRC_SIXLOWPAN
     dev->netdev.proto = GNRC_NETTYPE_SIXLOWPAN;
@@ -186,6 +185,8 @@ void at86rf2xx_tx_prepare(at86rf2xx_t *dev)
 {
     uint8_t state;
 
+    printf("%s():%d Incrementing pending_tx to %u\n", __func__, __LINE__,
+           dev->pending_tx + 1);
     dev->pending_tx++;
     /* make sure ongoing transmissions are finished */
     do {
diff --git a/drivers/at86rf2xx/at86rf2xx_netdev.c b/drivers/at86rf2xx/at86rf2xx_netdev.c
index 582d44c..b7573c3 100644
--- a/drivers/at86rf2xx/at86rf2xx_netdev.c
+++ b/drivers/at86rf2xx/at86rf2xx_netdev.c
@@ -624,7 +624,10 @@ static void _isr(netdev2_t *netdev)
             /* check for more pending TX calls and return to idle state if
              * there are none */
             assert(dev->pending_tx != 0);
+            printf("%s():%d Decrementing pending_tx to %u\n", __func__, __LINE__,
+           dev->pending_tx - 1);
             if ((--dev->pending_tx) == 0) {
+                printf("%s():%d => going IDLE\n", __func__, __LINE__);
                 at86rf2xx_set_state(dev, dev->idle_state);
             }
 ```

OlegHahm, 2016-04-28:
At least one TX IRQ seems to be missing. The only thing to further debug this for now would be using an oscilloscope to monitor the GPIO ping which signals the TX IRQ.

authmillenon, 2016-04-28:
I think we should open a separate issue for this and we fix this issue by deactivating the TX IRQs for lwIP for now. Do you agree?

OlegHahm, 2016-04-28:
Deactivating TX IRQs wouldn't work.

authmillenon, 2016-04-28:
Just realized that too :(

authmillenon, 2016-04-28:
Okay, then let's deactivate MLD per default until we figure this out. This at least limit the number of "fast transmissions" to 2 instead of 3, which the driver seems to be able to handle.

OlegHahm, 2016-04-30:
Just realized that I have the same problem with ccn-lite in my application. A possible workaround would be to always set the timer back to RX after some time (as a kind of a watchdog).

miri64, 2016-06-01:
#5404 is just a workaround so let's keep this one open

OlegHahm, 2016-06-01:
Really? The bug is somewhere in the driver.

miri64, 2016-06-01:
Is there an issue for that?

OlegHahm, 2016-06-01:
Good question - couldn't find one. Will take care of it.

OlegHahm, 2016-06-01:
done

kYc0o, 2016-07-11:
I think this issue can be postponed (as well as #5486), what do you think?

OlegHahm, 2016-07-13:
I'm afraid we have to.

kYc0o, 2016-07-13:
Well... let's do it...

miri64, 2016-07-13:
Well there is an issue for that now, so we can close this ;-)

PR #5404 (2016-05-31): lwip: introduce MLD as pseudo-module and deactivate by default
See https://github.com/RIOT-OS/RIOT/issues/5394#issuecomment-215378841

(Fixes #5394 in a not so nice way)
closed 2016-07-13 by miri64