RIOT-OS/RIOT #5406: xtimer_now() not monotonic

malosek, 2016-04-29:
Hello,
was trying to measure time of some ext events and found out that xtimer_now can go back in time (on msp430 platform).
Was checking msp TRMs and seems like if there is timer interrupt while maskable interrupts are disabled then the interrupt is lost (I need to recheck this, but it would have big implications on xtimer implementation).

Test code may looks like:
static void *blink_function (void *arg)
{
    xtimer_t xtimer;

    while (1)
    {
        kernel_pid_t me = thread_getpid();

        LED_RED_TOGGLE;

        uint32_t before= 0;
        uint32_t after = 0;

        before = xtimer_now();

        for (uint32_t i = 0; i < 100; i++) {
            __asm__("   nop");
            __asm__("   nop");
            __asm__("   nop");
            __asm__("   nop");
            __asm__("   nop");
            __asm__("   nop");
            __asm__("   nop");
            __asm__("   nop");
            __asm__("   nop");
            __asm__("   nop");
            __asm__("   nop");
            __asm__("   nop");
            __asm__("   nop");
            after = xtimer_now();

            if (after < before) {
                PRINTF("time shift before 0x%lX after 0x%lX\n", before, after);
            }
        }

        xtimer_set_wakeup(&xtimer, 1000000, me);
        thread_sleep();
    }
}

int main(void)
{
    PRINTF("You are running RIOT on a(n) %s board.\n", RIOT_BOARD);

    /* create blink thread */
    thread_create(blink_stack, sizeof(blink_stack),
                THREAD_PRIORITY_MAIN,
                CREATE_WOUT_YIELD | CREATE_STACKTEST,
                blink_function, NULL, blink_name);

    ps();

    return 0;
}

my output is:
RIOT MSP430 hardware initialization complete.
main(): This is RIOT!
You are running RIOT on a(n) ts430pm64 board.
        pid |        name |     stateQ | pri | stack ( used) | location
          1 |        idle |  pending Q |  15 |   128 (  100) | 19a4
          2 |        main |  running Q |   7 |   896 (  220) | 1100
          3 |       blink |  pending Q |   7 |   768 (  100) | 1480
time shift before 0x30BFEC6 after 0x30B0002
time shift before 0x30BFEC6 after 0x30B1755
time shift before 0x30BFEC6 after 0x30B2F25
time shift before 0x30BFEC6 after 0x30B46F4
time shift before 0x30BFEC6 after 0x30B5EC3
time shift before 0x30BFEC6 after 0x30B7693
time shift before 0x30BFEC6 after 0x30B8E6B
time shift before 0x30BFEC6 after 0x30BA63B
time shift before 0x30BFEC6 after 0x30BBE03

so apparently 16bits timer overflow was lost. Suppose #5338 has the same origin.
I'll try to reproduce it on mainline build with mspgcc as well, since this is some older code ported to TI compiler (v15.12.0.LTS) inside CCS - anyway expecting the same results there.

wbr
malo

kaspar030, 2016-04-30:
I've adapted your example to RIOT master. It is blinking since a couple of minutes now, and I don't get shifts so far. I'm on msb-430h, compiled with the RIOT docker toolchain.

Have you been running the exact code? If the shifts start at 0x30B0002, is it always after that time? Is that roughly at ~50sec after reboot?

> was trying to measure time of some ext events and found out that xtimer_now can go back in time (on msp430 platform).

> seems like if there is timer interrupt while maskable interrupts are disabled then the interrupt is lost

If you measured the time within ISRs or within irq_disable() ... irq_restore(), #5338 will be the reason.

malosek, 2016-04-30:
Hello,
Have been running this code exactly - measuring time in normal thread. I'm not next to computer but believe that  issue will appear with identical numbers each time running the code (will check that during the week).
With msp there are not nested nor pending interrupts, once when timer should trigger interrupt and interrupts are disabled, there is just flag set in timer module and that is all. Such condition can happen quite easy - for example when user timer is firing close to hw timer overflow.
This is just theory, but I'll continue to investigate.
Wbr
Malo

kaspar030, 2016-04-30:
> With msp there are not nested nor pending interrupts

AFAIK the platform supports both.

malosek, 2016-05-01:
Hello @kaspar030 ,

reg nested IRQs they are disabled by default (slau144j.pdf)
2.2.3.1 Interrupt Acceptance
The interrupt latency is 5 cycles ...
6. The SR is cleared. This terminates any low-power mode. Because the GIE bit is cleared, further
interrupts are disabled.

anyway you can always enable then within interrupt handler...

as for pending interrupts
2.2.3 Interrupt Processing
When an interrupt is requested from a peripheral and the peripheral interrupt enable bit and GIE bit are
set, the interrupt service routine is requested. Only the individual enable bit must be set for (non)-
maskable interrupts to be requested.

Could not find nothing more info about what is going to happen when maskable interrupts are enabled again - do you have some reference?

have feeling that issue is caused by interaction/overlap of user timer with timer overflow....

wbr
malo

malosek, 2016-05-02:
Hello @kaspar030 ,
Neither I could reproduce it on the RIOT master...
next period is "fixed" within _timer_callback properly.
uffff:)
anyway will continue to dig into it.
wbr
malo

malosek, 2016-05-03:
Hello,
The problem was related to my msp CPUX port, where I was saving status register after __disable_irq in thread_yield_higher (called from thread_sleep) so was saving and restoring the thread context with interrupts disabled...
After fixing that all works as should and in fact was just manifestation of #5338 .
Nice exercise anyway:) Thanks for you time @kaspar030 
wbr
malo

closed 2016-05-03 by malosek