RIOT-OS/RIOT #5407: Networking not working while gpio_init_int for button initialized

Nordzisko, 2016-04-29:
Hello guys,

I am working on STM32F3Discovery with Atmel AT86RF233 radio, trying to run microcoap server. Everything is working, while I dont add initialization for user button BTN_B1_PIN. 

Networking is working well, but when I add this 

`gpio_init_int(BTN_B1_PIN, GPIO_IN, GPIO_RISING, cb, (void*) 0);`

into microcoap_server main.c, the networking stop working. 
This is how function cb looks like:

` int flag_stav_2 = 0;

void cb(void *arg)
{
	if ( flag_stav_2 == 0 )	
	{
		LED5_ON;
		LED6_ON;	
		flag_stav_2 = 1;
	}
	else 
	{ 
		LED5_OFF;
		LED6_OFF;	
		flag_stav_2 = 0;
	}
}`

After adding that line interrupt is working, but networking don't. 

Do you know what can be an issue ? 
Thanks

PeterKietzmann, 2016-05-17:
Hi @Nordzisko! First of all, are you using the openlabs transceiver or something custom made? Just asking because of personal interest :-)...
The Atmel driver defines a default pin mapping when not defined by a board. As you can see [here](https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf2xx/include/at86rf2xx_params.h#L41) the chip select pin is `GPIO_PIN(0, 0)` which means PA0. On the stm32f3discovery board, `BTN_B1_PIN` lies on the same pin, compare [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/stm32f3discovery/include/board.h#L89). You could either change the default configuration of the driver or add a special configuration for your board, as done for boards with on-board transceivers, compare [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/samr21-xpro/include/board.h#L46). Obviously you can not use the same pin for two different things at the same time. All right?

Nordzisko, 2016-05-17:
Hey @PeterKietzmann ! 
Currently I am using Atmel ATZB-A-233-XPRO - http://www.atmel.com/tools/ATZB-A-233-XPRO.aspx, but later, I am planning to made some custom board. 

About the issue: I was aware of that situation with the same board pin for both functions, so in my file /boards/stm32f3discovery/include/board.h, I have added this: 

`#define AT86RF2XX_PARAMS_BOARD      {.spi = SPI_0, \
                                     .spi_speed = SPI_SPEED_5MHZ, \
                                     .cs_pin = GPIO_PIN(PORT_A, 4), \
                                     .int_pin = GPIO_PIN(PORT_B, 0), \
                                     .sleep_pin = GPIO_PIN(PORT_A, 1), \
                                     .reset_pin = GPIO_PIN(PORT_A, 2)}
`

So you had a good point in the answer, but I had it like this while opening this issue. 
Do you have any other idea, what can go wrong ? 
Thanks.

PeterKietzmann, 2016-05-17:
Hmm... Hard to say from afar. Did I get it correctly that communication works as expected with your above proposed configuration **but** once you try to initialize the user button as interrupt pin the driver (?!) stops working? And the user button works as expected without any networking? Could you try to initialize a different I/O line as interrupt? In addition, would you share a branch incl. application that you use for your experiments?

Nordzisko, 2016-05-19:
Hey,

Yes, you are right. Without the initialization of user button as interrupt, networking is working. 
And yes, user button is working as expected, when initialized as interrupt. 

Please, what do you mean by "initialize a different I/O line as interrup" ?

PeterKietzmann, 2016-05-19:
Instead of 

`gpio_init_int(BTN_B1_PIN, GPIO_IN, GPIO_RISING, cb, (void*) 0);`

try

`gpio_init_int(GPIO_PIN(PORT_C, 1), GPIO_IN, GPIO_RISING, cb, (void*) 0);`

(Didn't check if PC1 is free and available on that board)

Nordzisko, 2016-05-19:
Well, I do not have my workstation with RIOT code with me, but I suppose, that this suggestion will probably not work because if I would initialize interrupt pin as PORT_C.1, it will not do anything because I expect USER BUTTON to be connected on the physical level with the PORT_A.0 .

Am I missing something or this is good hypothesis ?

PeterKietzmann, 2016-05-20:
`gpio_init_int()` initializes "any" pin as external interrupt source. Thy pin may not be connected to a button but you can apply low-high or high-low transitions as well, to trigger an interrupt. With that proposal I wanted to figure out if your problem is still about pin conflicts or if there's something going wrong in the driver, especially looking at the ISR

PeterKietzmann, 2016-05-20:
PS: If you don't face problems with `GPIO_PIN(PORT_C, 1)` as an alternative to `BTN_B1_PIN`, I suppose we still overlooked something in the peripheral pin configurations.

PeterKietzmann, 2016-06-02:
@Nordzisko what's the latest?!

Nordzisko, 2016-06-02:
Hello, I am sorry for not being active for a while. 

I tried to initialize another pin with a external interrupt and networking was working. So I suppose, there is some incompatiblility in the driver and interrupt handler.

PeterKietzmann, 2016-06-03:
I still have the impression there must be a pin conflict. The interrupt handler isn't even executed cause the error already occurs after initialization. And it must be related to the initialization of **both**, pin and transceiver, cause both do work separately.
- do you connect your device with cables? Because your above proposed configuration includes pins not lying next to each other
- I'm running out of ideas. Can you share your code?
- In doubt, connect an external button to `PC1` ;-)

PeterKietzmann, 2016-06-03:
Do you have an other board available? For example stm32f**4**discovery?

Nordzisko, 2016-06-03:
I have now my code on my github page, the specific microcoap application with the interrupt is in the directory RIOT/examples/microcoap_interrupt. ( If there is any problem with viewing it, just contact me please, I am a newbie here on github :) )

Yes I have the radio connected using cables. 

I am sorry, I do not have any other board available.

PeterKietzmann, 2016-06-03:
@Nordzisko please notice the [Developement procedures](https://github.com/RIOT-OS/RIOT/wiki/Development-procedures) as well as the [git cheatsheet](https://github.com/RIOT-OS/RIOT/wiki/Git-cheatsheet). You don't need to do a PR but in general it really beneficial to keep your master branch even with the RIOT master and commit **your** changes in a new branch.

Anyway, I think I won't be able to go into details before next week.

Nordzisko, 2016-06-03:
@PeterKietzmann Thank you for the reflections, I was just trying to upload my code asap, so you can have a look at it. I will go through the references and consider reorganizing my branches. 

Don't worry about the time when you look at it, the project I was working on is already finished, so I just want to solve this problem out of pure curiosity.

PeterKietzmann, 2016-06-06:
I assume the interrupt pin is problematic

.int_pin = GPIO_PIN(PORT_B, 0)

because the board button lies on `GPIO_PIN(PORT_A, 0)`. IMO both are mapped to the same interrupt line which might lead to unwanted behaviour. The good thing is that you are flexible using jumper cables :-). Could you try to initialize the transceiver with: 

`.int_pin = GPIO_PIN(PORT_B, 0)`

Nordzisko, 2016-06-06:
As you can see (https://github.com/Nordzisko/RIOT/blob/Riot-Mine/boards/stm32f3discovery/include/board.h) , it is initialized as .int_pin = GPIO_PIN(PORT_B, 0) . Still no effect taken.

PeterKietzmann, 2016-06-06:
Sorry, that was a copy paste mistake. Please try with 

`.int_pin = GPIO_PIN(PORT_C, 3)`

Nordzisko, 2016-06-14:
Hey, I tried both (PORT_C,0) or (PORT_C,3) and in the both cases, not even networking was working.

PeterKietzmann, 2016-06-15:
Dammit! Im running out of ideas... But the button still works then? Fine, I will buy myself an stm32f3discovery so I can at least check the interrupt functionalities then.

Nordzisko, 2016-06-15:
Yes, Button was still working. If I will catch some new idea, I will inform you.

PeterKietzmann, 2016-06-23:
@Nordzisko my board arrived. I tested two setups that just differ in one detail. Note that I did not connect a transceiver.

Manual pin initialization like in the at86rf2xx driver:
1) SLEEP pin at PA1
2) RESET pin at PA2
3) INT pin at PB0 **AND** PC3
4) USR_BUTTN pin at PA0
5) SPI_0 at PA5-7
6) SPI_CS pin at PA4

Results:
- with PC3 in step 3) all pins worked as expected. GPIO outputs, GPIO interrupt lines (button and dedicated INT pin) as well as the SPI bus
- with PB0 in step 3) the interrupt functionality won't work after initialization of the USER_BUTTN in step 4) , as I described some time ago. Doing 4) before 3) prohibits the button to work correctly but the INT pin does

These are no new insights but if you carefully initialized and connected your device as I proposed above and you still face the same problems, the bug might not be related to the peripherals, even if it really creates the impression.

PeterKietzmann, 2016-08-04:
@Nordzisko are there any new insights?

Following [this](https://lists.riot-os.org/pipermail/devel/2016-July/004273.html) topic on our devel mailing list, @Adeel had problems with connecting an Atmel ATZB-**X**-233-XPRO. For me it seems like a problem with peripherals or peripheral configurations. Do you have an idea about the difference to the ATZB-**X**-233-XPRO or any good advice for Adeel  how to proceed?

adeelm87, 2016-08-04:
Thanks for your message on this thread @PeterKietzmann. You have tagged the wrong Adeel though :-).

I would like to add a little here about the differences between ATZB-X-233-XPRO and ATZB-A-233-XPRO. ATZB-A-233-XPRO only has a at86rf233 transceiver and the SPI pins on the external connector directly connect to the trasceiver. ATZB-X-233-XPRO on the other hand has an MCU and an at86rf233 trasceiver. The SPI pins on the external connector connect to the MCU. The MCU has an interface to the at86rf233 transceiver internally. I am not sure if this should be a problem.

Nordzisko, 2016-08-08:
Hello @PeterKietzmann  ! 
I am sorry, but this was school project and during the vacation, I do not have any access to the specific boards. 

@adeelm87 I would consider adding Makefile.dep as I did [https://github.com/Nordzisko/RIOT/tree/Riot-Mine/boards/stm32f3discovery](url) and add its reference to the Makefile.include file. Lookup for differencies in the folder I gave you link to, maybe you will find something :)

PeterKietzmann, 2016-08-09:
@adeelm87 how exactly do you connect to that "board"? I have no idea how this interface works but I could imagine that it causes trouble. Is the MCU meant to bypass the SPI signal somehow? RIOT provides a SPI driver for that radio and directly connecting from an stm32f4discovery worked. 
For me it seems a bit like the Atmel samr21 which has a samd21 CPU plus a transceiver on one chip which are internally connected via SPI .

miri64, 2016-10-18:
Is this still an issue and can someone please summarize what the reason is?

PeterKietzmann, 2016-10-19:
@Nordzisko we are currently preparing for the next RIOT release and it would be great to clarify this issue. Does it still persist on your site? IIRC I tested interrupt functionality as well as communication with an openlabs transceiver connected to the stm32f3discovery, with success. Would be great if you could report the latest news

Nordzisko, 2016-10-19:
@PeterKietzmann I do not have access to the devices, because my work was a part of school project, so I haven't had any progress since the last comment from me and I assume that I will not continue on this project, sorry.

PeterKietzmann, 2016-10-19:
@Nordzisko thanks for your feedback. It's quite hard to figure out what the bug is/was cause I can't reproduce. I will close this issue now because I’m not convinced this is an issue with RIOT or custom code. Please feel free to post back in here once you take on work again.

closed 2016-10-19 by PeterKietzmann