RIOT-OS/RIOT #5411: question: driver configuration in board.h file

toonst, 2016-05-03:
I'm working on a driver for the mcp2515. It's an CAN controller which uses SPI. As an example implementation I use the at86rf2xx driver, since it is similar also uses SPI.

I saw that for defining the number of devices used, the at86rf2xx_params array is used in `drivers/at86rf2xx/include/at86rf2xx_params.h`:
```
#define AT86RF2XX_PARAMS_DEFAULT    {.spi = AT86RF2XX_PARAM_SPI, \
                                     .spi_speed = AT86RF2XX_PARAM_SPI_SPEED, \
                                     .cs_pin = AT86RF2XX_PARAM_CS, \
                                     .int_pin = AT86RF2XX_PARAM_INT, \
                                     .sleep_pin = AT86RF2XX_PARAM_SLEEP, \
                                     .reset_pin = AT86RF2XX_PARAM_RESET}
/**@}*/

/**
 * @brief   AT86RF231 configuration
 */
static const at86rf2xx_params_t at86rf2xx_params[] =
{
#ifdef AT86RF2XX_PARAMS_BOARD
    AT86RF2XX_PARAMS_BOARD,
#else
    AT86RF2XX_PARAMS_DEFAULT,
#endif
};

```
So in the board file you can define a list of configurations, eg. in `boards/iotlab-m3/include/board.h`:
```
/**
 * @name Define the interface to the AT86RF231 radio
 *
 * {spi bus, spi speed, cs pin, int pin, reset pin, sleep pin}
 */
#define AT86RF2XX_PARAMS_BOARD      {.spi = SPI_0, \
                                     .spi_speed = SPI_SPEED_5MHZ, \
                                     .cs_pin = GPIO_PIN(PORT_A, 4), \
                                     .int_pin = GPIO_PIN(PORT_C, 4), \
                                     .sleep_pin = GPIO_PIN(PORT_A, 2), \
                                     .reset_pin = GPIO_PIN(PORT_C, 1)}

```
I'm wondering is this is the prefered way to define the configuration. I don't see the need for a default configuration, so to me it seems cleaner to define the configuration directly in the board.h file without a macro. Is there some other reason to define this configuration with a macro?
Thanks a lot!

PeterKietzmann, 2016-05-13:
Hi @toonst! On a board the configuration is fix, obviously. But the idea behind it was (IIRC) to have a default configuration in case you use an external transceiver on "any" board. E.g.the  [openlabs](http://openlabs.co/OSHW/Raspberry-Pi-802.15.4-radio) or [MRF24J40MA](http://www.microchip.com/wwwproducts/en/MRF24J40MA) transceiver with an stm32f4discovery. In that case the configuration is not fix and it is not strongly dependant on the board any more. You think that should be handled in an other way?

toonst, 2016-05-13:
Hi Peter! Unless you would want to define some parameters in `board.h` and some in `at86rf2xx_params.h` (I don't think you would want that), you could do it like this in `at86rf2xx_params.h`  :
```
/**
 * @brief   AT86RF231 configuration
 */
#ifndef AT86RF2XX_PARAMS_BOARD
static const at86rf2xx_params_t at86rf2xx_params[] =
{
   {
       .spi = SPI_0,
       .spi_speed = SPI_SPEED_5MHZ,
       .cs_pin = GPIO_PIN(0, 0),
       .int_pin = GPIO_PIN(0, 1),
       .sleep_pin = GPIO_PIN(0, 2),
       .reset_pin = GPIO_PIN(0, 3)
   }
};
#endif
```
And in `board.h`:
 ```
#define  AT86RF2XX_PARAMS_BOARD 1

static const at86rf2xx_params_t at86rf2xx_params[] =
{
   {
       .spi =...
       .spi_speed = ...
       .cs_pin = ...
       .int_pin = ...
       .sleep_pin = ...
       .reset_pin =  ...
   }
};
```
What do you think? It is a bit less flexible, but it does get rid of quiet some preprocessor macros.

PeterKietzmann, 2016-05-17:
Hi Toon. Honestly I don't really see the improvement. Is it because e.g. `AT86RF2XX_PARAM_SPI` is defined in at86rf2xx_params.h (even if not used) in case a different configuration from  board.h is present? I personally don't really care so much about preprocessor macros and IMO the current solution is quite logic. @thomaseichinger might have a more in-depth opinion about that.

toonst, 2016-05-17:
Hi Peter, OK that's fine. I wanted to learn the reasoning behind it but I understand now. Thanks for your comments!

PeterKietzmann, 2016-05-17:
@toonst I will close the issue now. If you need more in-depth feedback from one of the authors of the driver, feel free to reopen, mention and/or assign them to this issue

closed 2016-05-17 by PeterKietzmann