RIOT-OS/RIOT #5420: Off by one error in cbor_deserialize_float()

mdkrajnak, 2016-05-06:
At line 524 in the current master of cbor.c at:

https://github.com/RIOT-OS/RIOT/blob/master/sys/cbor/cbor.c

The function returns 4, but as it needs to account for the preceding byte that flags a float value this should be 5.

You can see the same pattern at lines 549-550 for cbor_deserialize_double() where it returns 9 instead of 8.

[cbor.zip](https://github.com/RIOT-OS/RIOT/files/252810/cbor.zip)

```
size_t cbor_deserialize_float(const cbor_stream_t *stream, size_t offset, float *val)
{
    if (CBOR_TYPE(stream, offset) != CBOR_7 || !val) {
        return 0;
    }

    unsigned char *data = &stream->data[offset];

    if (*data == CBOR_FLOAT32) {
        *val = ntohf(*(uint32_t *)(data + 1));
        return 4;
    }

    return 0;
}
```

authmillenon, 2016-05-06:
Would you be so nice to provide a pull request for that? It seems that you know pretty well where those bugs occure.

mdkrajnak, 2016-05-06:
I'd like being more helpful, but I'm not operating off of forked repository, I've just extracted the one file for standalone compilation. I could do a full checkout, change, and send a pull request, but I generally like to build projects before I submit a pull request and I'm not read to do a full build yet.

miri64, 2016-10-20:
Fix provided in #5975.

PR #5975 (2016-10-25): cbor: fix off-by-one error
Fixes #5420. I don't understand the unittests enough to see if they pick it up, and if yes, how to fix them. (or how to fix them that they *can* register such off by one errors, if they are not testing for it).
closed 2016-10-25 by x3ro