RIOT-OS/RIOT #5422: problem in build microcoap server for the cc2538dk

satishbabu2136, 2016-05-09:
i have the problem in build microcoap server for the cc2538dk. when i try to build the example the following error occured.

syalamarthi@syalamarthi:~/alljoyn/core/RIOT-2016.04/examples/microcoap_server$ make BOARD=cc2538dk
Building application "microcoap_server" for "cc2538dk" with MCU "cc2538".

"make" -C /home/syalamarthi/alljoyn/core/RIOT-2016.04/pkg/microcoap
rm -Rf /home/syalamarthi/alljoyn/core/RIOT-2016.04/examples/microcoap_server/bin/pkg/cc2538dk/microcoap
mkdir -p /home/syalamarthi/alljoyn/core/RIOT-2016.04/examples/microcoap_server/bin/pkg/cc2538dk/microcoap
/home/syalamarthi/alljoyn/core/RIOT-2016.04/dist/tools/git/git-cache clone "git://github.com/1248/microcoap.git" "9cb1dcda2182a8dca8483b230cda8b591a924c82" "/home/syalamarthi/alljoyn/core/RIOT-2016.04/examples/microcoap_server/bin/pkg/cc2538dk/microcoap"
+ local REMOTE=git://github.com/1248/microcoap.git
+ local SHA1=9cb1dcda2182a8dca8483b230cda8b591a924c82
++ basename git://github.com/1248/microcoap.git
+ local REMOTE_NAME=microcoap.git
+ local TARGET_PATH=/home/syalamarthi/alljoyn/core/RIOT-2016.04/examples/microcoap_server/bin/pkg/cc2538dk/microcoap
+ _check_commit 9cb1dcda2182a8dca8483b230cda8b591a924c82
+ git_cache cat-file -e '9cb1dcda2182a8dca8483b230cda8b591a924c82^{commit}'
+ git -C /home/syalamarthi/.gitcache cat-file -e '9cb1dcda2182a8dca8483b230cda8b591a924c82^{commit}'
Unknown option: -C
usage: git [--version] [--exec-path[=<path>]] [--html-path] [--man-path] [--info-path]
           [-p|--paginate|--no-pager] [--no-replace-objects] [--bare]
           [--git-dir=<path>] [--work-tree=<path>] [--namespace=<name>]
           [-c name=value] [--help]
           <command> [<args>]
+ git clone git://github.com/1248/microcoap.git /home/syalamarthi/alljoyn/core/RIOT-2016.04/examples/microcoap_server/bin/pkg/cc2538dk/microcoap
Cloning into '/home/syalamarthi/alljoyn/core/RIOT-2016.04/examples/microcoap_server/bin/pkg/cc2538dk/microcoap'...
remote: Counting objects: 172, done.
remote: Total 172 (delta 0), reused 0 (delta 0), pack-reused 172
Receiving objects: 100% (172/172), 34.41 KiB | 32 KiB/s, done.
Resolving deltas: 100% (92/92), done.
+ git -C /home/syalamarthi/alljoyn/core/RIOT-2016.04/examples/microcoap_server/bin/pkg/cc2538dk/microcoap checkout 9cb1dcda2182a8dca8483b230cda8b591a924c82
Unknown option: -C
usage: git [--version] [--exec-path[=<path>]] [--html-path] [--man-path] [--info-path]
           [-p|--paginate|--no-pager] [--no-replace-objects] [--bare]
           [--git-dir=<path>] [--work-tree=<path>] [--namespace=<name>]
           [-c name=value] [--help]
           <command> [<args>]
make[1]: *** [/home/syalamarthi/alljoyn/core/RIOT-2016.04/examples/microcoap_server/bin/pkg/cc2538dk/microcoap/.git-downloaded] Error 129
make: *** [/home/syalamarthi/alljoyn/core/RIOT-2016.04/examples/microcoap_server/bin/cc2538dk/microcoap.a] Error 2

kaspar030, 2016-05-09:
@satishbabu2136 Which version of git are you using? (```git --version```)

satishbabu2136, 2016-05-09:
it is fixed by updating the git by the version 2.2 to the 2.8.2

thanks for your support

closed 2016-05-09 by satishbabu2136