RIOT-OS/RIOT #5423: drivers/at86rf2xx: missing AT86RF2XX_OPT_TELL_RX_END

PeterKietzmann, 2016-05-09:
If I see it correctly the option `AT86RF2XX_OPT_TELL_RX_END` is available on the device and provided to the netdev2 interface. Actually it was intended to be enabled in the [init function](https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf2xx/at86rf2xx.c#L107) but there is no driver implementation that really sets the option on the device (compare [here](https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf2xx/at86rf2xx_getset.c#L337)). Or am I blind?

PeterKietzmann, 2016-05-10:
I was wrong. There is no option for a decidated RX_END interrupt on that device. It is `TRX_END`. The function `at86rf2xx_set_option()` is just used to set the appropriate flag in the netdev option field.

closed 2016-05-10 by PeterKietzmann

authmillenon, 2016-05-10:
Pity, thought it might be the cause of https://github.com/RIOT-OS/RIOT/issues/5394

OlegHahm, 2016-05-10:
Hm, I don't think so. TRX_END is just not triggered for some reason.

PeterKietzmann, 2016-05-10:
Not? Everything I mentioned above was just by looking at the code, nothing tested. Will look into it

PeterKietzmann, 2016-05-10:
@OlegHahm how did you determine that? I took the current master and ran gnrc_networking on two boards. and as it seems (to me) `TRX_END` *is* triggered cause I arrive [here](https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf2xx/at86rf2xx_netdev.c#L620) and the netdev2 callback function is called.

PeterKietzmann, 2016-05-10:
If I see it correctly it must be to be able receiving packets. netdev2 is still new to me...

OlegHahm, 2016-05-10:
Sorry, I was imprecise: it is not *always* triggered, leading to problems with LWIP and in some other corner cases. The problem seems to happen if you send several packets back to back.

PeterKietzmann, 2016-05-10:
Any ideas about how often this happens, after how many packets and which packet size? I tried the following 3-4 times without drops. I used gnrc_networking on two nodes. The "receiver" *without* an initialized UDP server for not getting the whole pkt_dump outputs but with a note of the event [here]. Sending 1000 small UDP packets without a delay never showed less then 1000 received packets. Ok, there might be other packets from neighbour discovery or so, but mostly I receiver even some more packets then 1000 in the mac layer.

authmillenon, 2016-05-10:
`gnrc_networking` is "save" in that regard, because even with 0 delay between packets this delay will not be handed to the driver due to IPC slow-down. Try `tests/lwip` with or without https://github.com/RIOT-OS/RIOT/pull/5404 instead. Without https://github.com/RIOT-OS/RIOT/pull/5404, it sends out 3 packets rapidly in the very beginning (MLR, RS and NS) and after that you are not able to receive anything any more on this radio (even on driver level), because `pending_tx` never reaches 0 (because one TX_END event gets lost, allegedly because of a missing IRQ). With https://github.com/RIOT-OS/RIOT/pull/5404 you don't have this, because MLD get's deactivated -> now MLR -> only 2 rapid packets -> everything is fine.

authmillenon, 2016-05-10:
Ah small test application that boosts out packets on the driver level might be of help here.

PeterKietzmann, 2016-05-10:
>this delay will not be handed to the driver due to IPC slow-down

I don't completely understand that. Didn't we conclude from our network-stack measurements that processing times through the stack are times faster that the actual over-the-air time? With that in mind I'm not sure if the problem might be IPC-slow-down. Or I just don't understand it... Anyway, I will look into your proposed test but probably not today.

authmillenon, 2016-05-10:
Okay from the top: GNRC uses IPC calls to hand a packet to the driver, lwIP does not and accesses (at least for sending) the driver directly via function calls. This means that GNRC is basically slowed down by the IPC by a few μs, while lwIP doesn't get delayed at all (that's also the reason why the slope of the lwIP tx traversal time is much less steaper than of the other two stacks). That's what I meant with IPC-slow-down. It has nothing to do with over-the-air time ;-)

PeterKietzmann, 2016-05-10:
Hm. Actually I think it *has* something to do with over-the-air time. Can you comment on my following opinion please? This is what I've assumed to happen when two packets are sent back to back (focusing on GNRC)

- send() packet 1 from application
- IPC and processing through network stack
- in the driver, check if device is busy ([compare])(https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf2xx/at86rf2xx.c#L193)
- device is not busy, so directly load the transceiver FIFO, call execute() and leave the rest to the hardware
- the driver returns just then *but* the hardware may take way longer to send the data (over the air)
- send() packet 2 from application
- IPC and processing through network stack
- now again in the (blocking) function of the driver, the device is still busy under the assumption that radio transmission is slower than packet processing (what we've verified with our measurements)
- so we already have the next frame to be sent prepared in the driver and were just waiting until the device is not busy anymore. Once the transceiver is idle again, we can directly send the next frame

So, as long as data transmission takes longer than processing and the driver is blocking while busy, I think the "IPC-slowdown" should not affect the frame rate at the output of the transceiver (and thus at the receiver as well). 

The `RX_DONE` callback overhead is something that should apply for all network-stacks that implement netdev2, if I see it correctly.

Did I get something wrong? Or did that maybe indicate some hints to the bug you're referring to?

OlegHahm, 2016-05-10:
I agree with your assessment, @PeterKietzmann. However, I still don't understand where and how the IRQ gets lost, but didn't find the time yet to monitor the corresponding pin with an oscilloscope yet.

authmillenon, 2016-05-10:
That has nothing to do with the bug I described: Every time you send something [this counter](https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf2xx/at86rf2xx.c#L189) is incremented and supposed to be decremented [here](https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf2xx/at86rf2xx_netdev.c#L627). The problem is for more than 2 rapidly send packets (i.e. no IPC in-between) the `_isr()` function is not actually called at least once. This means, that the `tx_pending` counter never reaches 0, never bringing the device back to idle state, meaning the device isn't able to receive anymore. The counter was introduced to not accidentally read packets that are supposed to be send as received from the buffer.

authmillenon, 2016-05-10:
But the way you described it for *your* use-case is right so far ;-)

PeterKietzmann, 2016-05-10:
But this is a problem on the sender side, right? I thought your initial problem was about receiving when they've been sent too fast, but that is not the fact if I get you correctly. Sorry, still didn't find time to look into your issues listed above

OlegHahm, 2016-05-10:
The problem is that a sender will not know that it is done sending, when the bug occur and hence, never return to RX.

PeterKietzmann, 2016-05-10:
Thanks, now I got it. Will hopefully find some time to think about it the next days