RIOT-OS/RIOT #5432: native: valgrind fails

OlegHahm, 2016-05-12:
Can anybody tell me what
```
==28122== Conditional jump or move depends on uninitialised value(s)
==28122==    at 0x804CA72: ??? (cpu/native/tramp.S:85)
==28122==  Uninitialised value was created by a stack allocation
==28122==    at 0x410D35B: setcontext (in /usr/lib32/libc-2.23.so)
==28122== 
==28122== Use of uninitialised value of size 4
==28122==    at 0x804CA73: ??? (cpu/native/tramp.S:87)
==28122==  Uninitialised value was created by a stack allocation
==28122==    at 0x410D35B: setcontext (in /usr/lib32/libc-2.23.so)
==28122== 
==28122== Jump to the invalid address stated on the next line
==28122==    at 0x0: ???
==28122==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==28122== 
==28122== 
==28122== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==28122==  Bad permissions for mapped region at address 0x0
==28122==    at 0x0: ??
```
is trying to tell me when I run my RIOT application with valgrind? It doesn't happen everytime, but often when I create a somewhat bigger native network or after some time.

Kijewski, 2016-05-12:
Aren't lines 81 and 83 in the wrong order in tramp.S? Could you try swapping them? To me it looks like this might fail if the instance gets bombarded with IO signals, e.g. flood ping.

OlegHahm, 2016-05-12:
```diff
diff --git a/cpu/native/tramp.S b/cpu/native/tramp.S
index 70d808d..97b354b 100644
--- a/cpu/native/tramp.S
+++ b/cpu/native/tramp.S
@@ -78,9 +78,9 @@ _native_sig_leave_tramp:
     call swapcontext
     addl $8, %esp
 
-    call irq_enable
-
     movl $0x0, _native_in_isr
+
+    call irq_enable
     popal
     popfl
```
didn't help.

OlegHahm, 2016-05-12:
With #2071 merged these problems seem to be gone.

kYc0o, 2016-07-11:
@OlegHahm can you confirm that this issue is gone with that PR merged?

OlegHahm, 2016-07-11:
I think so, but #2071 has not yet been merged due to the missing ARM support.

kYc0o, 2016-07-11:
Ok it seems complicated to get it for the next release so let's postpone it again... hopefully the last time! :)

OlegHahm, 2016-11-10:
#2071 has been merged.

closed 2016-11-10 by OlegHahm