RIOT-OS/RIOT #5434: Lake of learning resources

ideadx, 2016-05-12:
Please put some links on learning resources. Features and everything else is great but coming from js background, I need some reading material to understand how things are put to gather. Whats on here in read me on project is not helping much.

Please share book or some other material that help understand things.
here is some inspirations: https://github.com/rwaldron/johnny-five

This project has real shot on to merge as leading IoT platform but needs a solid docs...
Thanks.

authmillenon, 2016-05-12:
Hi,
first of all, welcome to the community. We know, that our doc is not the best for people coming from the high-end and [we are working on it](https://github.com/RIOT-OS/RIOT/labels/DocTF). Have you checked out http://doc.riot-os.org/ already? Please note however, that this kind of message is more suitable for the [devel mailinglists](http://lists.riot-os.org/mailman/listinfo/devel). As this goes basically in the same direction as #2137, I would vote to close this issue. Any objections?

authmillenon, 2016-05-12:
Side question: what excactly do you mean by "learning resources"? On RIOT itself or the tools we are using?

ideadx, 2016-05-13:
@authmillenon mainly RIOT itself. something like: https://laravel.com/docs/5.2

acochrane, 2016-05-30:
I have also found the docs a little impenetrable.
My first hurdle was figuring out that the usage of 'addr', as explained in every beginner tutorial I found, doesn't seem to exist in current RIOT.

I think some slight updates to your wiki section in the examples will help immensely.
Learning will and progress will occur quicker when your tutorials work exactly as described.

I was wondering if it's possible do to a pull request on the wiki.

I would like update the default example tutorial at the very least.

acochrane, 2016-06-02:
I just realized the howto linked on the wiki is the readme in the example and thus am feeling sheepish. I will have a go at updating it based on my recent experience.

Lotterleben, 2016-10-07:
Hi @acochrane, I just stumbled across your comment– I was wondering if you'd gotten around to updating the howto (not quite sure which one you mean exactly) with your experiences? :) Also, if you see anything in the Wiki that you feel is incomplete, you can just edit the page. If you're unsure or want to let people know about the changes you've made, you can send an e-mail to the RIOT mailing list before/after editing.

miri64, 2019-08-10:
We have the [tutorials](https://github.com/RIOT-OS/Tutorials) now which we also update in irregular intervals, so I think this issue is fixed

closed 2019-08-10 by miri64