RIOT-OS/RIOT #5436: z1 ram overflow - microCoAP_server

AppleInside, 2016-05-12:
Hi everyone, today i tried to build the microcoap_server example on my Zolertia z1 board, but the compiling process ends with a "RAM overflow of 832 bytes" error.
There is a way to fix this problem by myself?

Thanks a lot! 👍

malosek, 2016-05-13:
Hello @AppleInside ,
it is possible to run coap server on MSP with networking stack (6lowpan node), you just need to be little more RAM careful.
The basic rule is that if you are missing RAM somewhere then you have to shave of few bytes from the places when it is not used:)
Do you try plan to have 6lowpan host or node?

Anyway check flags from gnrc minimal example
CFLAGS += -DGNRC_PKTBUF_SIZE=512 -DGNRC_IPV6_NC_SIZE=0

wbr
malo

AppleInside, 2016-05-13:
Hi @malosek,
i'm trying to build the sample version of the coap-server (the one included in the _**RIOT/examples/microcoap_server**_ directory).
I don't know which files "shave" for have a lighter ram footprint, moreover in the microcoap's related Makefile there is indicated the list of the board that have "insufficient memory" for it, so maybe there is no way to make it compatible.

br,
Alessandro

aabadie, 2017-07-07:
Digging this up: I would propose to use the nanocoap example instead. I built it locally and it fits on the Z1:
```
$ make BOARD=z1 -C examples/nanocoap_server/

[...]

   text	   data	    bss	    dec	    hex	filename
  51990	     34	   7084	  59108	   e6e4	.../RIOT/examples/nanocoap_server/bin/z1/nanocoap_server.elf
```

closed 2017-07-07 by aabadie