RIOT-OS/RIOT #5438: RIOT and Cooja - $ make cooja

AppleInside, 2016-05-13:
Hi all,
I'm following this guide for simulating a 6LowPan network on cooja.
https://github.com/RIOT-OS/RIOT/wiki/RIOT-and-Cooja

But when i try to download cooja (in the RIOT path) with the command "make cooja", it doesn't work.
Is there anyone who has already done this?

Thank u all.

authmillenon, 2016-05-13:
Hi @AppleInside,
please consider using one of the maliling lists for questions? Here you will only reach active developers, while e.g. over the [devel mailinglist](https://lists.riot-os.org/mailman/listinfo/devel) you also reach developers less active and over the [users mailinglist](https://lists.riot-os.org/mailman/listinfo/users) you also reach other users (though from experience its not as active).

Cheers,
Martine

AppleInside, 2016-05-13:
Thank you @authmillenon for the tip, i'll do it asap!

br,
Alesssandro

miri64, 2016-10-18:
In case you never got an answer: I realize now, that Cooja (https://github.com/RIOT-OS/RIOT/pull/1921) was never merged. I'm deleting that article because of that.

kYc0o, 2016-10-18:
But does it mean that RIOT and Cooja does not work? In my opinion this is a MUST...

miri64, 2016-10-18:
> But does it mean that RIOT and Cooja does not work?

Not at the moment, but I don't see any reason against it.

> In my opinion this is a MUST...

Feel free to adapt #1921 :-)

closed 2016-10-18 by miri64