RIOT-OS/RIOT #5443: core: msg: msg_send_receive() when receiver is not waiting

OlegHahm, 2016-05-16:
It seems there was a reason for #879 after all: if the receiving thread B is not waiting, but sending thread A uses `msg_send_receive()` A's `thread_t::wait_data` is overwritten, because it is used for both: storing the sent message until B gets into receiving mode **and** the reply.

A simple show case to illustrate this is https://github.com/OlegHahm/miniature-dangerzone/tree/master/msg_snd_rcv

A case where this bug shows in practice is `gnrc_netapi_get/set()` if the receiver's queue is full.

OlegHahm, 2016-05-16:
I wonder if checking for the receiver's state inside `msg_send_receive()` would be tolerable.

OlegHahm, 2016-05-16:
Something like:
```diff
diff --git a/core/msg.c b/core/msg.c
index 7bbb2cd..7d2a144 100644
--- a/core/msg.c
+++ b/core/msg.c
@@ -212,6 +216,17 @@ int msg_send_receive(msg_t *m, msg_t *reply, kernel_pid_t target_pid)
 {
     assert(sched_active_pid != target_pid);
     unsigned state = irq_disable();
+
+    thread_t *target = (thread_t*) sched_threads[target_pid];
+    if (target->status != STATUS_RECEIVE_BLOCKED) {
+        int res;
+        if ((res =_msg_send(m, target_pid, true, state)) >= 0) {
+            res = msg_receive(reply);
+        }
+        irq_restore(state);
+        return res;
+    }
+
```

authmillenon, 2016-05-16:
Wouldn't that make `msg_send_receive()` very insecure?

authmillenon, 2016-05-16:
Wouldn't it be better to `SEND_BLOCK` when target is blocked?

OlegHahm, 2016-05-16:
> Wouldn't that make msg_send_receive() very insecure?

Why?

> Wouldn't it be better to SEND_BLOCK when target is blocked?

What do you mean by `SEND_BLOCK`?

authmillenon, 2016-05-16:
>> Wouldn't that make msg_send_receive() very insecure?
>
> Why?

Because if the receiver isn't receive-blocked the message will get lost.

>> Wouldn't it be better to SEND_BLOCK when target is blocked?
> 
> What do you mean by SEND_BLOCK?

This state https://github.com/RIOT-OS/RIOT/blob/master/core/include/thread.h#L55

OlegHahm, 2016-05-16:
I still don't understand, why should the message get lost and which thread would you set to `THREAD_SEND_BLOCKED`? The caller of `msg_send_receive()` calls `_msg_send()` with `block == true` in both versions.

authmillenon, 2016-05-16:
> I still don't understand, why should the message get lost

Because there is no alternative path for `if (target->status != STATUS_RECEIVE_BLOCKED) {`. If the target is not receive blocked the function does nothing, right?

> and which thread would you set to `THREAD_SEND_BLOCKED`? The caller of msg_send_receive() calls _msg_send() with block == true in both versions.

The caller of `msg_send_receive()` I would set to `THREAD_SEND_BLOCKED`, in for that else case and unblock it, as soon as `(target->status != STATUS_RECEIVE_BLOCKED)` is the case (by the scheduler).

authmillenon, 2016-05-16:
Wait ... shouldn't it be `(target->status == STATUS_RECEIVE_BLOCKED)`.

OlegHahm, 2016-05-16:
> Because there is no alternative path for if (target->status != STATUS_RECEIVE_BLOCKED) {. If the target is not receive blocked the function does nothing, right?

No, if the receiver is `STATUS_RECEIVE_BLOCKED` `msg_send_receive()` will do the same as before, if not it will send and receive separately.

authmillenon, 2016-05-16:
Ahhhh, sorry, then I just missread the code >.<

PR #5444 (2016-05-31): core: msg_send_receive if receiver is not ready
If the receiving thread is not waiting, the sending thread will copy its `wait_data` into the reply.

Fixes #5443.
closed 2016-05-31 by miri64