RIOT-OS/RIOT #5458: Problems with BR example

aabadie, 2016-05-23:
Hi,
I tried to configure a BR following the example provided [here](https://github.com/RIOT-OS/RIOT/tree/master/examples/gnrc_border_router) and using ethos. But the prefix 2001:db8::/64 is not propagated in my network.
I tested it using a samr21-xpro board with gnrc_networking (on /dev/ttyACM1) for the node in my network and another samr21-xpro for the BR accessible via /dev/ttyACM0 (also tried with an iotlab-m3 with /dev/ttyUSB1).

These are the steps I performed:
1. flash gnrc_networking example on the samr21-xpro network node, open a serial terminal on /dev/ttyACM1
2. build uhcpd and ethos
3. flash gnrc_border_router on the samr21-xpro used as BR
4. start the network using the following command
<pre>
$ sudo ~/RIOT/dist/tools/ethos/start_network.sh /dev/ttyACM0 tap0 2001:db8::/64
</pre>
which outputs the following:
<pre>
net.ipv6.conf.tap0.forwarding = 1
net.ipv6.conf.tap0.accept_ra = 0
----> ethos: sending hello.
----> ethos: activating serial pass through.
----> ethos: hello reply received
uhcp_client(): timeout waiting for reply
uhcp_client(): sending REQ...
uhcp_client(): timeout waiting for reply
uhcp_client(): sending REQ...
uhcp_client(): timeout waiting for reply
uhcp_client(): sending REQ...
</pre>
5. running ifconfig on the BR returns:
<pre>
ifconfig
Iface  6   HWaddr: 16:76  Channel: 26  Page: 0  NID: 0x23
           Long HWaddr: 36:32:48:33:46:d8:96:76 
           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
           AUTOACK  CSMA  MTU:1280  HL:64  6LO  RTR  IPHC  
           Source address length: 8
           Link type: wireless
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::3432:4833:46d8:9676/64  scope: local
           inet6 addr: ff02::1:ffd8:9676/128  scope: local [multicast]
           
Iface  7   HWaddr: 00:21:d3:60:5a:8d 
           
           MTU:1500  HL:64  RTR  RTR_ADV  
           Source address length: 6
           Link type: wired
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::221:d3ff:fe60:5a8d/64  scope: local
           inet6 addr: ff02::1:ff60:5a8d/128  scope: local [multicast]
           inet6 addr: ff02::2/128  scope: local [multicast]
           inet6 addr: fe80::2/64  scope: local
           inet6 addr: ff02::1:ff00:2/128  scope: local [multicast]
</pre>
6. ifconfig on the other node (with gnrc_networking):
<pre>
Iface  7   HWaddr: 6c:a2  Channel: 26  Page: 0  NID: 0x23
2016-05-23 11:47:12,301 - INFO #            Long HWaddr: 5a:5a:65:62:05:8a:ec:a2 
2016-05-23 11:47:12,308 - INFO #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2016-05-23 11:47:12,313 - INFO #            AUTOACK  CSMA  MTU:1280  HL:64  6LO  RTR  IPHC  
2016-05-23 11:47:12,317 - INFO #            Source address length: 8
2016-05-23 11:47:12,319 - INFO #            Link type: wireless
2016-05-23 11:47:12,325 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2016-05-23 11:47:12,330 - INFO #            inet6 addr: fe80::585a:6562:58a:eca2/64  scope: local
2016-05-23 11:47:12,336 - INFO #            inet6 addr: ff02::1:ff8a:eca2/128  scope: local [multicast]
</pre>

The procedure was tested with either latest [master](https://github.com/RIOT-OS/RIOT/commit/29badd7d581ed96ce3e65a95537f7f6af4a3fad8) and [2016.04](https://github.com/RIOT-OS/RIOT/tree/2016.04-branch).
The build host is Ubuntu 16.04.

Hope this is clear. I can give more information if needed.

aabadie, 2016-05-23:
I'll try to analyze the problem better this afternoon.

cc @kaspar030

authmillenon, 2016-05-23:
Hi, just tried it again and for me it still works:

```
$ sudo dist/tools/ethos/start_network.sh /dev/ttyACM0 tap0 2001:db8::/64
net.ipv6.conf.tap0.forwarding = 1
net.ipv6.conf.tap0.accept_ra = 0
----> ethos: sending hello.
----> ethos: activating serial pass through.
----> ethos: hello reply received
uhcp_client(): timeout waiting for reply
uhcp_client(): sending REQ...
got packet from fe80::5067:82ff:fefa:e04c port 51860
uhcp: push from fe80::5067:82ff:fefa:e04c:51860 prefix=2001:db8::/64
gnrc_uhcpc: uhcp_handle_prefix(): configured new prefix 2001:db8::5841:2049:9619:203a/64
ifconfig
ifconfig
Iface  6   HWaddr: 20:3a  Channel: 26  Page: 0  NID: 0x23
           Long HWaddr: 5a:41:20:49:96:19:20:3a 
           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
           AUTOACK  CSMA  MTU:1280  HL:64  6LO  RTR  RTR_ADV  IPHC  
           Source address length: 8
           Link type: wireless
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::5841:2049:9619:203a/64  scope: local
           inet6 addr: ff02::1:ff19:203a/128  scope: local [multicast]
           inet6 addr: 2001:db8::5841:2049:9619:203a/64  scope: global
           inet6 addr: ff02::2/128  scope: local [multicast]
           
Iface  7   HWaddr: 00:21:d3:60:5a:8d 
           
           MTU:1500  HL:64  RTR  RTR_ADV  
           Source address length: 6
           Link type: wired
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::221:d3ff:fe60:5a8d/64  scope: local
           inet6 addr: ff02::1:ff60:5a8d/128  scope: local [multicast]
           inet6 addr: ff02::2/128  scope: local [multicast]
           inet6 addr: fe80::2/64  scope: local
           inet6 addr: ff02::1:ff00:2/128  scope: local [multicast]
```

Did you compile `uhcpd`? Make sure it is running (it seems like `uhcpd` isn't running on your machine, that's why I'm asking)

authmillenon, 2016-05-23:
Pinging the node in the network is also possible:

```
$ ping6 -c 3 2001:db8::5841:2644:b81c:2dfe
PING 2001:db8::5841:2644:b81c:2dfe(2001:db8::5841:2644:b81c:2dfe) 56 data bytes
64 bytes from 2001:db8::5841:2644:b81c:2dfe: icmp_seq=1 ttl=63 time=66.6 ms
64 bytes from 2001:db8::5841:2644:b81c:2dfe: icmp_seq=2 ttl=63 time=44.6 ms
64 bytes from 2001:db8::5841:2644:b81c:2dfe: icmp_seq=3 ttl=63 time=44.1 ms

--- 2001:db8::5841:2644:b81c:2dfe ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 44.119/44.713/45.366/0.566 ms
```

aabadie, 2016-05-23:
Thanks for the reply @authmillenon.
On the BR the prefix is not propagated : on the wireless interface, I can't see any global ip starting with 2001:db8 prefix.
uhcpd is running:
```bash
ps -elf | grep uhcpd
0 S root     29169 29134  0  80   0 -  1090 -      15:16 pts/20   00:00:00 /home/aabadie/RIOT/dist/tools/uhcpd/bin/uhcpd tap0 2001:db8:: 64
```

authmillenon, 2016-05-23:
Okay, I'll try to reproduce on my Ubuntu 16.04 machine at home again. As I said: I don't see this behavior on Arch (on my laptop).

aabadie, 2016-05-23:
I also have strange warnings when I build (forgot to mention them... sorry):
```
/arm-none-eabi/lib/armv6-m/libc_nano.a(lib_a-nano-reallocr.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv6-m/libc_nano.a(lib_a-nano-msizer.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
```
Maybe my toolchain is not setup correctly (the upgrade from 15.10 to 16.04 may have broken it)

aabadie, 2016-05-23:
> As I said: I don't see this behavior on Arch (on my laptop).

Which is a pretty good news ! Thanks.

aabadie, 2016-05-23:
@authmillenon, I tried again the set up on another computer running ubuntu 14.04. It's working. So I'm now trying to find where is the problem with ubuntu 16.04.

I'm interested in the result of your home experiment.

kYc0o, 2016-07-11:
Do you have news on this issue?

aabadie, 2016-07-11:
> Do you have news on this issue?

Works with 2016.04 release of RIOT but didn't tried recently with master.

kYc0o, 2016-07-26:
Please try asap if you can, I'll move this issue to the next release.

miri64, 2016-10-17:
@aabadie does this issue still exist?

aabadie, 2016-10-17:
No, the warning problem was fixed and the BR issue is another problem (see #5539). Closing this one.

closed 2016-10-17 by aabadie

kamejoko80, 2017-06-09:
Hello,

I see that after running this script it blocks the Linux console.
Is there any way to run script start_network.sh as a background service?