RIOT-OS/RIOT #5466: pkg/micro-ecc: micro-ecc cannot be built for SAMR21-XPRO board

wentaoshang, 2016-05-26:
micro-ecc patches in RIOT-OS adds the dependency on hardware rng, which is not available on SAMR21-XPRO board (and many others).  RIOT-OS should fall back to the original software rng in micro-ecc code if the target board does not have periph-hwrng support.

OlegHahm, 2016-05-27:
I am by no means a crypto expert, but still I wonder if it makes sense to base a crypto library on a software PRNG without a good source for seeding it. In other words, shouldn't we find a solution for https://github.com/RIOT-OS/RIOT/issues/3221 first, before we change this?

wentaoshang, 2016-05-27:
I'm not a security expert, either. So I don't know how well the rng in uECC is implemented... But if the board does not have hardware rng support, what other choice do we have except using software rng?

gebart, 2016-05-27:
@wentaoshang the challenge is to find innovative ways to gather entropy for the PRNG and use a cryptographically secure PRNG for the random numbers. (though I am not a cryptographer either)

wentaoshang, 2016-05-27:
I'd like to take a practical approach: why not use the tinymt32 implementation in RIOT-OS as a starting point? Is it too weak to be used for ECC?

d00616, 2016-05-27:
Without hardware RNG there must be implemented a very good pseudo RNG initialized by entropy from various sources. I have no knowledge to do this in a cryptographic save way. I'm not a crypto expert.

The major question is how to get entropy good enough for cryptography. How to get this fast to do cryptographic operations a short time after powering on. If this solved a pseudo RNG can used for faster generation of randomness.

@wentaoshang tinymt32 may a good solution for an pseudo RNG for key sizes up to 128 Bit. Have you FIPS tested it?

wentaoshang, 2016-05-27:
No I haven't done any testing. Not sure how to do "FIPS test"...

d00616, 2016-05-27:
Take a look here: https://wiki.archlinux.org/index.php/Rng-tools

kaspar030, 2016-05-27:
> I'd like to take a pragmatic approach: why not use the tinymt32 implementation in RIOT-OS as a starting point? Is it too weak to be used for ECC?

The problem is that with the same seed value, PRNGs spit out the same numbers in the exact same sequence. So while using tinymt32 as RNG would be OK for starting, we still need to find a way to initially seed it after reboot, in case no HW RNG is available.

wentaoshang, 2016-05-31:
@kaspar030 After a few days of research, I couldn't find a way to seed the PRNG on the boards with no hardware rng. There are rumors that a lot of ARM manufactures actually include HW rng on the chip. But not sure how to make use of that.

So another solution is to use `uECC_sign_deterministic` API. It follows the standard practice defined in RFC6979, which is designed specifically for devices with little or no source of entropy. It simply uses the hash of the private key to generate the random number `k` in ECDSA. Generating the private key still requires strong source of entropy. Therefore we need to generate the key on some other device then copy it over to the IoT board with not HWRNG (which makes sense to me).

This API is pretty easy to use. But I still have problem with linking: gcc must resolve all external references when it prepares the .a file for uECC, but `hwrng_read` is never defined (even though it's never used in the code, either). Is it possible to add some macro tricks so that the 0002 patch is applied only if HWRNG is supported by the board?

OlegHahm, 2016-05-31:
Maybe `FEATURES_PROVIDED` can be used somehow.

mtausig, 2016-07-26:
Introducing a weak RNG is definietely not a good solution. 
But what could be done, is make micro-ecc at least semi-usable without a HWRNG, since a lot of functions does not need it anyway (like signature verification and deterministic signing). One could create a patch which removes all functions requiring random numbers when no hwrng is available.

mtausig, 2016-07-26:
By the way, @wentaoshang: Have a look at https://github.com/RIOT-OS/RIOT/pull/4989, which implements the hwrng for a samr1-xpro board. It is not merged (and won't be in the near future), but it works and you could use it in your project.

PR #6204 (2017-01-05): pkg/micro-ecc: support boards without hwrng feature
With the old micro-ecc patch, applications that use micro-ecc cannot be compiled on boards without hwrng feature because `hwrng_read` is not defined. This new patch will insert `hwrng_read` and `riot_hwrng` functions only if `FEATURE_PERIPH_HWRNG` is defined. This allows boards with no HWRNG to use deterministic ECDSA signing. This PR will fix #5466.

Some related discussions are in #5914 and #5466.
closed 2017-01-05 by OlegHahm