RIOT-OS/RIOT #5467: link local address vanishes

smlng, 2016-05-27:
In short, I observed that link-local address vanishes when using RIOT with module 'gnrc_ipv6_default' instead of 'gnrc_ipv6_router_default' if (and only if) more than 1 router is in the network.

In detail, my test setup is as follows:

- RasPi with openlabs transceiver acts as boarder router running `radvd` advertising an IPv6 prefix (`FDxy:...::/64` something)
- 2 RIOT nodes running gnrc_networking example w/o RPL, and one of them using  module 'gnrc_ipv6_default' instead of 'gnrc_ipv6_router_default'
- hence, there are two routers (RasPi + 1 RIOT node) in the network and 1 plain RIOT node
- both RIOT nodes generate a (global) IPv6 address using the prefix in the RA send by RasPi+radvd
- issue: the RIOT node with module 'gnrc_ipv6_default' loses its link local, shortly thereafter

Ifconfig output from shell of the latter node:
```
2016-05-27 09:53:12,552 - INFO # > ifconfig
2016-05-27 09:53:12,556 - INFO # Iface  7   HWaddr: 62:9e  Channel: 16  Page: 0  NID: 0x2804
2016-05-27 09:53:12,558 - INFO #            Long HWaddr: 5a:55:42:45:c8:bb:e2:9e 
2016-05-27 09:53:12,565 - INFO #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2016-05-27 09:53:12,570 - INFO #            AUTOACK  CSMA  MTU:1280  HL:64  6LO  IPHC  
2016-05-27 09:53:12,574 - INFO #            Source address length: 8
2016-05-27 09:53:12,576 - INFO #            Link type: wireless
2016-05-27 09:53:12,584 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2016-05-27 09:53:12,587 - INFO #            inet6 addr: fe80::5855:4245:c8bb:e29e/64  scope: local
2016-05-27 09:53:12,588 - INFO #            

2016-05-27 09:53:14,188 - INFO # > ifconfig
2016-05-27 09:53:14,191 - INFO # Iface  7   HWaddr: 62:9e  Channel: 16  Page: 0  NID: 0x2804
2016-05-27 09:53:14,194 - INFO #            Long HWaddr: 5a:55:42:45:c8:bb:e2:9e 
2016-05-27 09:53:14,200 - INFO #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2016-05-27 09:53:14,205 - INFO #            AUTOACK  CSMA  MTU:1280  HL:255  6LO  IPHC  
2016-05-27 09:53:14,208 - INFO #            Source address length: 8
2016-05-27 09:53:14,211 - INFO #            Link type: wireless
2016-05-27 09:53:14,217 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2016-05-27 09:53:14,222 - INFO #            inet6 addr: fe80::5855:4245:c8bb:e29e/64  scope: local
2016-05-27 09:53:14,229 - INFO #            inet6 addr: fd9c:4bb5:28fb:0:5855:4245:c8bb:e29e/64  scope: global
2016-05-27 09:53:14,230 - INFO #            

2016-05-27 09:53:15,381 - INFO # > ifconfig
2016-05-27 09:53:15,385 - INFO # Iface  7   HWaddr: 62:9e  Channel: 16  Page: 0  NID: 0x2804
2016-05-27 09:53:15,387 - INFO #            Long HWaddr: 5a:55:42:45:c8:bb:e2:9e 
2016-05-27 09:53:15,394 - INFO #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2016-05-27 09:53:15,399 - INFO #            AUTOACK  CSMA  MTU:1280  HL:255  6LO  IPHC  
2016-05-27 09:53:15,403 - INFO #            Source address length: 8
2016-05-27 09:53:15,405 - INFO #            Link type: wireless
2016-05-27 09:53:15,413 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2016-05-27 09:53:15,418 - INFO #            inet6 addr: fd9c:4bb5:28fb:0:5855:4245:c8bb:e29e/64  scope: global
2016-05-27 09:53:15,418 - INFO #
```

I'm currently debugging this, but I assume this happens when the node receives a RA from the other RIOT node. Because, when I shutdown the other RIOT node (with module `gnrc_ipv6_router_default`) and RasPi is the only router in the network, link-local address does not vanish in RIOT.

smlng, 2016-05-27:
I made some more tests, the issue arises only if I run my tests repeatedly, rebooting only the RIOT node with `gnrc_ipv6_default` but not the other one with `gnrc_ipv6_router_default`. 

The problem is that the RIOT router node still has an (old) ncache entry of the other RIOT node. So, the RIOT node sends a NS including its ARO to the RIOT router node. The latter one handles the ARO and finds an existing entry in its ncache, hence it answers with a NA including ARO with `SIXLOWPAN_ND_STATUS_DUP`. When the RIOT node receives the RA it has to remove the address from its interface, in that case its link-local address, but it does not generate an alternative one!

TL;DR: duplicate address detection strikes, but RIOT does not generate an alternative address, it simply deletes the address from the interface and in worst case no communication is possible.

PeterKietzmann, 2016-05-27:
See also #3052

smlng, 2016-05-27:
Oh and by the way, this is not link local address specific -> global addresses will also disappear if still present in the ncache of a router node.

gebart, 2016-05-27:
Good analysis! 
Should not the router during duplicate address detection see that the link layer address of the node asking for duplicate address detection is the same as in the neighbor cache? 
I don't know the ND standards very well so I don't know what it says regarding the above.

miri64, 2016-05-29:
Yes it should will add it to my long list of issues in ND that need an overhaul ;)

miri64, 2017-11-06:
Fixed by #7925

miri64, 2017-11-27:
Confirmed this to be the case multiple times over the last week. (#7925 is merged)

closed 2017-11-27 by miri64