RIOT-OS/RIOT #5486: at86rf2xx: lost interrupts

OlegHahm, 2016-06-01:
The introduction of https://github.com/RIOT-OS/RIOT/pull/5261 revealed that - if packets are sent back to back - it seems to happen, that interrupts indicating the successful transmission of a packet (i.e. TXEND) are somehow lost, which leads to the fact that the state machine gets stuck. This bug can be reproduced on iotlab-M3 or samr21-xpro using lwip with MLD enabled (as described in #5394).

jfischer-phytec-iot, 2016-06-01:
Also observed the same behavior while porting kw2xrf to netdev2 #5469,
so I do not blocking it, https://github.com/RIOT-OS/RIOT/pull/5469/files#diff-b4ec7aa2ee5925942c4e7e2f506b695cR109

jia200x, 2016-07-25:
@OlegHahm  as pending_tx is an uint8_t, at some point this variable overflows and becomes 0 again (=> crash due to assert).

```
1469461021.682398;m3-53;Tx_pending: 250
1469461021.682609;m3-53;[at86rf2xx] EVT - TX_END
1469461021.683507;m3-53;[at86rf2xx] TX NO_ACK
1469461021.696359;m3-53;Exec
1469461021.716527;m3-53;[at86rf2xx] EVT - RX_END
1469461021.746462;m3-53;Exec
1469461021.803949;m3-53;[at86rf2xx] EVT - RX_END
1469461021.808569;m3-53;Exec
1469461021.830631;m3-53;[at86rf2xx] EVT - RX_END
1469461021.866592;m3-53;Exec
1469461021.897651;m3-53;[at86rf2xx] EVT - RX_END
1469461021.924615;m3-53;Exec
1469461021.943853;m3-53;[at86rf2xx] EVT - RX_END
1469461021.976520;m3-53;Exec
1469461022.013008;m3-53;[at86rf2xx] EVT - RX_END
1469461022.032714;m3-53;Exec
1469461022.046508;m3-53;Tx_pending: 0
1469461022.047852;m3-53;0x800ecf1
1469461022.048052;m3-53;*** RIOT kernel panic:
1469461022.048225;m3-53;FAILED ASSERTION.
1469461022.048389;m3-53;
1469461022.050336;m3-53;    pid | name                 | state    Q | pri | stack ( used) | base       | current    
1469461022.051341;m3-53;      - | isr_stack            | -        - |   - |   512 (  132) | 0x20000000 | 0x200001c8
1469461022.053436;m3-53;      1 | idle                 | pending  Q |  15 |   256 (  132) | 0x20000488 | 0x20000504 
1469461022.056384;m3-53;      2 | main                 | bl mutex _ |   7 |  1536 (  512) | 0x20000588 | 0x20000a8c 
1469461022.058366;m3-53;      3 | ot_thread            | running  Q |   6 |  3072 ( 1876) | 0x200011a0 | 0x20001cfc 
1469461022.059306;m3-53;        | SUM                  |            |     |  5376 ( 2652)
1469461022.059545;m3-53;
1469461022.059726;m3-53;*** halted.
1469461022.059898;m3-53;
```

(I added "Exec" DEBUG msg to at86rf2xx_tx_exec ).

This seems to happen when there's an incoming packet just after transmission. Maybe one of this solutions?:

1. Give space for missing interrupts (e.g declare a timeout for each transmission, and only if the driver can't handle all interrupts on time).
2. Shorten the starting time of ACK (flag AT86RF2XX_XAH_CTRL_1__AACK_ACK_TIME). This way there might be more chances for the interrupt to be called.

For me this only happens when there's a high traffic of packets.
Cheers

jia200x, 2016-07-25:
I tried setting the AT86RF2XX_XAH_CTRL_1__AACK_ACK_TIME flag to 1, but didn't work...

miri64, 2016-08-04:
This issue also now causes large IPv6 datagrams not to be received anymore. See https://github.com/RIOT-OS/Release-Specs/issues/21#issuecomment-236606806

OlegHahm, 2016-08-04:
> This issue also now causes large IPv6 datagrams not to be received anymore.

Isn't the problem rather caused by *sending* large datagrams? Otherwise I don't see the connection.

miri64, 2016-08-04:
Well the receiver get's stuck, too. So there is definitely a connection.

OlegHahm, 2016-08-04:
Have you tested with receiving only? If this is also leads to a situation where the device's getting stuck, that may indicated another problem (which might be related to this issue or not).

miri64, 2016-08-04:
Yes, both sender and receiver have >0 counters pinging multiple >500 byte-sized frames with <=100 ms delay.

OlegHahm, 2016-08-04:
Now I'm even more confused. How can a receiving only device have a *TX* counter > 0?

miri64, 2016-08-04:
Because it's sending replies.

OlegHahm, 2016-08-04:
Then I repeat my question: Isn't the problem rather caused by *sending* large datagrams?

miri64, 2016-08-04:
Ah okay, that's what you wanted to infer. Yes, but the TX_END interrupt get's lost => no decrementation of the tx counter.

miri64, 2016-11-04:
Maybe, just maybe this was fixed by https://github.com/RIOT-OS/RIOT/pull/5244?

miri64, 2017-06-27:
Sadly #7115 did not fix this problem

aabadie, 2017-06-29:
> Sadly #7115 did not fix this problem

@miri64 since #7115 was in fact broken, maybe it will work with #7275

miri64, 2017-06-29:
For the record: this can be tested on `iotlab-m3` (not sure if this bug exists on `samr21-xpro` or other boards) with the following patch and the `tests/lwip` application:

```diff
diff --git a/drivers/at86rf2xx/at86rf2xx.c b/drivers/at86rf2xx/at86rf2xx.c
index df8b711..e17565b 100644
--- a/drivers/at86rf2xx/at86rf2xx.c
+++ b/drivers/at86rf2xx/at86rf2xx.c
@@ -139,6 +139,7 @@ void at86rf2xx_tx_prepare(at86rf2xx_t *dev)
     uint8_t state;
 
     dev->pending_tx++;
+    printf("dev->pending_tx (inc): %u\n", dev->pending_tx);
 
     /* make sure ongoing transmissions are finished */
     do {
diff --git a/drivers/at86rf2xx/at86rf2xx_netdev.c b/drivers/at86rf2xx/at86rf2xx_netdev.c
index a9452c8..2581226 100644
--- a/drivers/at86rf2xx/at86rf2xx_netdev.c
+++ b/drivers/at86rf2xx/at86rf2xx_netdev.c
@@ -560,6 +560,7 @@ static void _isr(netdev_t *netdev)
                 at86rf2xx_set_state(dev, dev->idle_state);
                 DEBUG("[at86rf2xx] return to state 0x%x\n", dev->idle_state);
             }
+            printf("dev->pending_tx (dec): %u\n", dev->pending_tx);
 
             DEBUG("[at86rf2xx] EVT - TX_END\n");
 
diff --git a/tests/lwip/Makefile b/tests/lwip/Makefile
index 58753af..5ccecad 100644
--- a/tests/lwip/Makefile
+++ b/tests/lwip/Makefile
@@ -12,6 +12,7 @@ BOARD_INSUFFICIENT_MEMORY := airfy-beacon nrf6310 nucleo32-f031 nucleo32-f042 \
 
 # including lwip_ipv6_mld would currently break this test on at86rf2xx radios
 USEMODULE += lwip lwip_ipv6_autoconfig lwip_sock_ip lwip_netdev
+USEMODULE += lwip_ipv6_mld
 USEMODULE += lwip_udp lwip_sock_udp
 USEMODULE += lwip_tcp lwip_sock_tcp
 USEMODULE += ipv6_addr
```

If `dev->pending_tx (dec): 0` is not printed after all packets were transmitted (there is already some transmission at the start, so make sure you see everything from `reset`), then the bug still exists.

DipSwitch, 2018-08-12:
Please look at my write-up under https://github.com/RIOT-OS/RIOT/pull/7276#issuecomment-412254790

OlegHahm, 2018-09-17:
Does anyone know if this bug does still exist?

PeterKietzmann, 2019-08-19:
@jia200x is this issue still present?

jia200x, 2019-08-19:
Unfortunately I've still seen some lost interrupts while stress testing :/

maribu, 2022-09-16:
@jia200x Could you do another stress test? I only read the title of the issue and it looks like the move to `sys/event` for the ISR handling should have solved it, doesn't it?

jia200x, 2022-09-16:
I would be confident on just closing this one, since it's impossible that it triggers "lost interrupts" anymore (first the message is not there anymore as a result of https://github.com/RIOT-OS/RIOT/pull/16748, and second events cannot be lost).
However, there are still race conditions when sending replies too fast (see https://github.com/RIOT-OS/RIOT/pull/11256#issuecomment-571091533).
That bug is hard to fix with `netdev` flow, but it gets fixed with https://github.com/RIOT-OS/RIOT/pull/16535)

jia200x, 2022-09-16:
so, let's just close this one.

closed 2022-09-16 by jia200x