RIOT-OS/RIOT #5487: RFC: shell: split WLAN/WPAN specific options into own command

miri64, 2016-06-01:
`ifconfig` is starting to get really complicated and I'm thinking about splitting some radio specific options into their own command (similar to Linux' `iwconfig` and `iwpan`). Opinions?

jfischer-phytec-iot, 2016-06-01:
``iwpan`` for ``802.15.4`` and ``iwconfig`` for ``802.11`` ?

OlegHahm, 2016-06-01:
iwconfig was deprecated alredy 5 years ago - we should rather go for `iw` then.

jfischer-phytec-iot, 2016-06-01:
then ``iwpan`` for ``802.15.4`` and ``iw`` for ``802.11``

miri64, 2016-06-01:
Yes, for example.

jfischer-phytec-iot, 2016-06-01:
What about not ieee std devices like cc110x?

OlegHahm, 2016-06-01:
Do we really need to split this up for different radio technologies? Most properties (channel, TX power, network identifier) should be rather similar for most technologies.

miri64, 2016-06-01:
Yes most, but as #5469 already shows, there are some that can even be very specific for vendors.

OlegHahm, 2016-06-01:
But that's (as you wrote yourself) vendor specific, not specific to the technology and would prefer to keep the number of shell commands small. Even for 802.11 chips on Linux you'll find a broad variety of available options, but (fortunately) only one tool.

jfischer-phytec-iot, 2016-06-01:
I am on iwpan, will add it to  #5485

miri64, 2016-06-01:
@jfischer-phytec-iot can you please introduce it in a seperate PR and just adapt it in #5485? It will make the review more focused this way.

miri64, 2016-06-01:
Also, when push comes to shove I'm for `iw` instead of `iwpan` ;-).

jfischer-phytec-iot, 2016-06-01:
@miri64 ok

PeterKietzmann, 2019-02-21:
#5750 was closed 2 years ago. Since then, there were no efforts for a change. Furthermore, we continued caring about `ifconfig`. Personally I think we could close this issue (maybe set memo label) and admit we stay with `ifconfig`. @miri64 any objections?

miri64, 2019-02-21:
Seeing the amount of new information added to ifconfig, I'm still of the opinion that we actually should factor out certain options that are only of importance for certain devices. However, since this is a low priority (at least to me), we should keep this issue open as a reminder for that / opportunity for a newcomer to implement that. What do you think @PeterKietzmann?

PeterKietzmann, 2019-02-21:
Personally I don't think this will ever happen, although it might be worth it. But I'm fine leaving the issue open. However, I don't think this is something that should be done by a newcomer.

miri64, 2019-09-25:
#12295 gave some new precedence to this. @jia200x @benpicco what do you think?

benpicco, 2019-09-25:
#12171 will add even more radio commands to `ifconfig`.

I too agree that it would be a good idea to have separate commands for layer 2 and layer 3.

jia200x, 2019-09-25:
+1. Indeed!

closed 2020-04-28 by stale[bot]