RIOT-OS/RIOT #5494: cpu: Implement thread_arch_isr_stack_usage() for missing platforms

miri64, 2016-06-01:
#5491 introduced the capability to print the stack-usage of the ISR. Currently, this feature is only provided  for Cortex-M platforms and needs to be provided for all other platforms:

- [ ] `atmega2560`/`atmega_common`
- [ ] `msp430_common`
- [ ] `lpc2387`/`arm7_common`
- [ ] `qemu-i386`

kYc0o, 2016-07-26:
Postponing for the next release.

madokapeng, 2018-04-11:
Hi, I am a new comer to RIOT OSS. 
This issue is assigned to me. I realize it was created a year ago. any updates?

miri64, 2018-04-11:
First of all, a warm welcome to the RIOT community @madokapeng.

This isn't really my strong suit, but I would suggest to look into the current implementation in `cortexm_common` and go from there from the platform you are working on.

Out of interest: who assigned you to this issue?

madokapeng, 2018-07-25:
@miri64 
Hi, I got this issue from codetriage email. I should not say "assign to me" :)
what is the procedure to solve issues?
is just by looking at open Issues list and pick up one?

Thanks

closed 2019-09-10 by stale[bot]