RIOT-OS/RIOT #5497: void * vs char * vs uint8_t *

kaspar030, 2016-06-02:
We do use the three inconsistently.

I would like to collect facts on what we know about each other, then make an informed decision on when to use what, then make the uses consistent.

kaspar030, 2016-06-02:
I'll start collecting what I know:

- C allows use of any type pointer as ```void *``` and the other way around without casting
- C using ```void*``` and ```char*``` has different semantics than using other pointer types regarding to strict aliasing rules. As uint8_t is just a typedef, is it equivalent to the others in that regard?

- ```char c; if (c == 0xff) ...``` is always false, which leads to unexpected bugs

Subjective stuff:

- using uint8_t when handling byte data feels more natural compared to ```char*```
- using ```void*``` improves readability a lot as no casting is needed
- dealing with a mixed ```char *``` and ```uint8_t``` codebase sucks because of needed casts

kaspar030, 2016-06-02:
- standard C code usually uses ``` char* ``` for everything string related (printf, strcmp/len/tok/...) and ```void*``` for everything where buffers are supplied (read/write/memcpy/memset/...)

Kijewski, 2016-06-02:
Fun fact: char != signed char != unsigned char. Whether `c == 0xff` is implementation defined.

kYc0o, 2016-06-02:
I think we should stick to the common sense and try to be consistent with the standard C coding style.
For me a `char*` is always a pointer to string related (character buffers and so on), `uint8_t*` to integer buffers or addresses and void* to function pointers, undefined return type callbacks... I don't know what is the current use of such pointer types on RIOT though...

kaspar030, 2016-06-02:
> Fun fact: char != signed char != unsigned char. Whether c == 0xff is implementation defined.

They revised that in C99:

```
C99 N1256 draft 6.2.5/15 "Types" has this to say about the signed-ness of type char:

    The implementation shall define char to have the same range, representation, and behavior as either signed char or unsigned char.

and in a footnote:

    CHAR_MIN, defined in <limits.h>, will have one of the values 0 or SCHAR_MIN, and this can be used to distinguish the two options. Irrespective of the choice made, char is a separate type from the other two and is not compatible with either.
```

So char for us (--std=c99) is either signed char or unsigned char, or is it really a third type despite having the same range as signed or unsigned char?

kYc0o, 2016-06-02:
do you have a more concrete example for that case?

kaspar030, 2016-06-02:
```
[kaspar@booze src]$ cat test.c
#include <stdio.h>

int main(void)
{
    char c = 0xff;
    if (c==0xff) {
        printf("c==0xff\n");
    } else {
        printf("c!=0xff\n");
    }
    return 0;
}
[kaspar@booze src]$ gcc -std=c99 -Wall test.c -o testx
[kaspar@booze src]$ ./testx
c!=0xff
[kaspar@booze src]$
```

```-pedantic``` would catch the assignment, though.

kYc0o, 2016-06-02:
yeah I understand the behaviour, the thing is that I cannot see how `char*` can be affected by such statement... you think that could affect a pointer that is in the upper or lower limits of the value?

miri64, 2016-06-02:
My usage of these types of *parameters* is atm:

* `char *`: strings (and only strings),
* `void *`: any buffer, type, thing that can be anything, I usually use that to type input or output data, but I also use
* `uint8_t *` for byte-arrays, especially if in the function I iterate over the buffer or if I edit parts of it.

kaspar030, 2016-06-02:
> how char* can be affected by such statement...

Ah, I mean for example if we'd use char* for incoming serial bytes, and wait for a control character (like, 0xff), this would need not always obvious casting when comparing to one of the "characters" the pointer points to.

miri64, 2016-06-02:
Just to make my conventions a little clearer (and I argue that at least in my work I'm pretty consistent with that): unless the type is 100% clear (e.g. I expect an IPv6 header) I use `void *`. The *only* exception for code-readability and to not having to need to store a temporary variable is when I do byte-related work on the data (e.g. iteration for checksums) in the function. Of the `c == 0xff` I did not know before, but I'm not sure I either I ran into any problems with that with `uint8_t *`. `char *` I only use in string related contexts (e.g. `ipv6_addr_to_str()`).

kaspar030, 2016-06-02:
> Of the |c == 0xff| I did not know before, but I'm not sure I either I
> ran into any problems with that with |uint8_t *|.

Doesn't affect uint8_t as 0xff is a perfectly valid value there.

kaspar030, 2016-06-02:
> unless the type is 100% clear (e.g. I expect an IPv6 header) I use |void *|.

Do you mean the expected type within the function or at the call site?
e.g., for UART, SPI, ... the callee always expects a byte array, but the
caller might send any type.

miri64, 2016-06-02:
>> unless the type is 100% clear (e.g. I expect an IPv6 header) I use |void *|.
>
> Do you mean the expected type within the function or at the call site?
e.g., for UART, SPI, ... the callee always expects a byte array, but the
caller might send any type.

Both, but primarily on the caller side. A case were it even isn't that clear for the callee would be `netdev_driver_t::get()`

miri64, 2016-06-02:
You do monospace by using ````` btw ;-)

kaspar030, 2016-06-02:
> You do monospace by using ````` btw ;-)

does it work when replying via email: ```foobar```?

miri64, 2016-06-02:
No, doesn't seem like it.

kaspar030, 2016-06-02:
> Both, but primarily on the caller side.

So whenever a buffer (which is equivalent to byte array) is expected, we'd use ```void*```, even if within the function we iterate over it? That would basically mean we use ```uint8_t``` as type for byte arrays but ```void*``` to pass them around, unless we're 100% sure that the caller will pass a definately untyped byte array, in which case we'd use ```uint8_t*``` for the parameter type?

LudwigKnuepfer, 2016-06-02:
I would opt for keeping all three around (and use them reasonably), too.

LudwigKnuepfer, 2016-06-02:
PS:
If anything I'd lose the `void *` - explicit casting is more reasonable in general.

miri64, 2016-06-02:
> If anything I'd lose the `void *` - explicit casting is more reasonable in general.

:-1: in any case I use it anything else doesn't make sense.

> So whenever a buffer (which is equivalent to byte array) is expected, we'd use void*, even if within the function we iterate over it? 

Huh? That was exactly the opposite of what I said, nor do I propose anything. I use `uint8_t *` when a byte-array is expected by the caller because e.g. it want's to iterate over it.

> That would basically mean we use uint8_t as type for byte arrays but void* to pass them around, unless we're 100% sure that the caller will pass a definitely untyped byte array, in which case we'd use uint8_t* for the parameter type?

For the first part I agree, but I'm not quite sure what we need to be "100% sure". When the situation arises, one look into the function's code makes clear, why I used a `uint8_t` at that point.

miri64, 2016-06-02:
Oh btw, just remember another thing I generally use `uint8_t *` for: link-layer addresses.

kaspar030, 2016-06-02:
> Huh? That was exactly the opposite of what I said, nor do I propose anything.

I know. ;) But you also said that you'd make the decision primarily based on the caller side expected type, which is contradicting. I thought we could clear that up.

miri64, 2016-06-02:
I reject your reality and substitute my own ;-).

No but seriously, I thought it was clear: When the type from the caller can be expected to be anything, than I use `void *`. That's what I wanted to say, but maybe I wasn't clear enough :dancer:

kaspar030, 2016-06-02:
I feel we're getting somewhere:

- use ```char*``` for strings and only for strings, because C + libc do it that way

- use ```uint8_t``` as type for untyped byte buffers, but use ```void *``` to pass them around. ```uint8_t``` because we're dealing with bytes and not characters, ```void*``` to avoid unnecessary casting

- use ```uint8_t*``` to pass "typed" byte buffers, e.g., link-layer addresses, where it avoids unnecessary temporary variable

- break those guidelines wherever reasonable ;)

miri64, 2016-06-02:
Is not wanting to convert all the usages of `msg.content.ptr` a reasonable reason to break these guidelines? Because I made multiple efforts to do so, but its just too much ^^.

kaspar030, 2016-06-02:
> Is not wanting to convert all the usages of msg.content.ptr a reasonable reason to break these guidelines? Because I made multiple efforts to do so, but its just too much ^^.

;) I'll try to find some sed magic.

LudwigKnuepfer, 2016-06-02:
> * use uint8_t as type for untyped byte buffers, but use void * to pass them around. uint8_t because we're dealing with bytes and not characters, void* to avoid unnecessary casting

Maybe I don't understand what you mean with "untyped byte buffers", but to me a byte buffer is typed - the type is byte, i.e. ` uint8_t` . If some "other side" knows that this buffer contains serialized data that is of type X an explicit cast to type X is a good thing in my book.

miri64, 2016-06-02:
I think what they means is stuff like the internall buffer of `pktbuf` or thread stacks, that need to have a type, because `void` arrays are not a thing

miri64, 2016-06-03:
I would add our results to the coding conventions if we are okay.

kaspar030, 2016-06-03:
yup! I'll close this issue. Thanks everyone!

miri64, 2016-06-03:
[Done](https://github.com/RIOT-OS/RIOT/wiki/Coding-conventions#types). Please check if you are alright with the wording, if not change to your liking.

OlegHahm, 2016-06-03:
Not even 24h for such a discussion seems to be a bit short.

kaspar030, 2016-06-03:
> Not even 24h for such a discussion seems to be a bit short.

Really, the change seems bigger than it is. All code but three printf's
compiled and worked with just the type change. The rest of the PR is
cosmetic.

LudwigKnuepfer, 2016-06-04:
> > Not even 24h for such a discussion seems to be a bit short.
> Really, the change seems bigger than it is. All code but three printf's
compiled and worked with just the type change. The rest of the PR is
cosmetic.

I agree with what @OlegHahm said and it's not about the PR but about the principle of passing `void *` around for buffers. I would discourage passing `void *` pointers around for generic memory buffers as this masks their (otherwise detectable by static analysis) erroneous/unintended use.

LudwigKnuepfer, 2016-06-04:
PS:
Please compare https://www.securecoding.cert.org/confluence/display/c/EXP36-C.+Do+not+cast+pointers+into+more+strictly+aligned+pointer+types

miri64, 2016-06-04:
Mh, I don't think we are speaking about the same kind of buffers. In the scope of this issue I think of buffers, where no sensible thing can be assumed about the type API-wise (see `msg.content.ptr` e.g.). In all of these cases I'm talking, they are however always marked either by a generic type field (again see `msg.content.ptr` with `msg.type`) or have also a length given with them (e.g. `memcpy()`), so I don't see how alignment or overflow issues can happen that static code analysis could not catch.

kaspar030, 2016-07-30:
> I would discourage passing void * pointers around for generic memory buffers as this masks their (otherwise detectable by static analysis) erroneous/unintended use.

I think we're talking about functions like ```memcpy()```. The idea is that wherever the pointer points to arbitrary data, like with memcpy or for a container holding an arbitrary pointer type, ```void *``` is the way to go, as it reduces otherwise necessary casting.

Stargateur, 2017-11-08:
@kaspar030 Hello, just reading randomly your thread.

`uint8_t` is not a byte it's a octet, `char` is a byte, this is important because your are talking about C compliant code. In C, byte != octet, char could be greater than 8 bits, but `uint8_t` if defined has exactly 8 bits. Plus, `uint8_t` is not a character type so strict aliasing exception with character type theoretically don't work with `uint8_t`.

However a lot of machine have a byte where CHAR_BIT is 8. But on some implementation this is not true.

liuziangexit, 2018-07-01:
@Stargateur i agree. in some embedded platform, a byte is 32 bit

kaspar030, 2018-07-01:
@Stargateur @liuziangexit of course you're right, C defines ```CHAR_BIT >= 8```.

Is anyone trying to use RIOT on such a platform?

Stargateur, 2018-07-01:
@kaspar030 I just talk about semantic, if `uint8_t` is available, `CHAR_BIT` would be 8. But keep in mind that `uint8_t` do not allow you to [alias](https://stackoverflow.com/q/98650/7076153) other type contrary to `char`.

liuziangexit, 2018-07-01:
yes, the CPP standard says you can only use the uchar and char to access an
obj, otherwise is UB(include using union, which is legal in C99 but illegal
in CPP

kaspar030, 2019-08-19:
> But keep in mind that `uint8_t` do not allow you to [alias](https://gist.github.com/jibsen/da6be27cde4d526ee564) other type contrary to `char`.

I thought "uint8_t" doesn't *have* to be a character type. It is when typedef'ed to "(signed) char", right?

Stargateur, 2019-08-19:
> > But keep in mind that `uint8_t` do not allow you to [alias](https://gist.github.com/jibsen/da6be27cde4d526ee564) other type contrary to `char`.
> 
> I thought "uint8_t" doesn't _have_ to be a character type. It is when typedef'ed to "(signed) char", right?

Yes but you will rely on implemented behavior, in my projet I avoid that, and personally I think it's a very bad way to implement `uint8_t`, this disallow a lot of optimization.

kaspar030, 2019-08-19:
> Yes but you will rely on implemented behavior, in my projet I avoid that, and personally I think it's a very bad way to implement `uint8_t`, this disallow a lot of optimization.

Just to be clear, it only disallows optimization in case uint8_t is *not* a typedef'ed char?

Stargateur, 2019-08-22:
> > Yes but you will rely on implemented behavior, in my projet I avoid that, and personally I think it's a very bad way to implement `uint8_t`, this disallow a lot of optimization.
> 
> Just to be clear, it only disallows optimization in case uint8_t is _not_ a typedef'ed char?

no that the contrary, that will not allow optimisation, if `uint8_t` is a typedef to `char`, the compiler will assume the data can alias anything, so it will disallow optimization on some case. Like a function that take a pointer to structure and a pointer to `char`, the compiler assume both could be alias, so if `uint8_t` is a typedef to `char` the same rule will apply. That why I don't like this way of implementation. See https://gcc.gnu.org/bugzilla/show_bug.cgi?id=66110, we can see that gcc make a bad decision but as always they stick with it.

You can see https://stackoverflow.com/questions/57259126/why-does-the-rust-compiler-not-optimize-code-assuming-that-two-mutable-reference, Rust should assume nothing alias on most case too. But does not currently.

maribu, 2023-05-17:
Looking at the code base and recent changes, there seems to be a consensus now:

- Use `void *` as return type / argument type for opaque data (such as data to transmit, store, read, encode as base64, ...) - which is consistent with the standard C lib and POSIX
- Use `char *` for strings only
- Use `uint8_t []` for network addresses

If anyone disagrees, please reopen.

closed 2023-05-17 by maribu