RIOT-OS/RIOT #5503: Mixed peripheral support

chrysn, 2016-06-03:
I see the need for different peripherals of the same kind to be usable simultaneously in RIOT. These could be, for example (the first 3-4 I have actual use cases for):

* The generic ARM SWO output debug exposed as a (monodirectional) UART for stdout
* Virtual UARTs that go to a pipe or a multiplexer (for example, Telit cellular chips can multiplex their UART into one data and one AT-command interface)
* Semihosted peripherals (eg. using the shell on devices w/o a connected or even present UART)
* ADCs attached via i2C
* GPIO pins attached via shift registers
* UARTs attached via USB
* Bit banging SPI (yikes)

Much (but not all) of this could be solved by adding an actual intermediate layer, but that would leave library implementors the choice of incurring a general performance and complexity penalty, or risking that the library won't work in all situations. (Would, for example, stdio switch to an arbitrary socket-ish interface? Would any particular SPI user switch their CS pin from GPIO to abstract-gpio?)

The solution I'm proposing roughly consists of the following, all of which can easily be made optional at preprocessor level, so imagine all of the following prefixed by "If mixed peripherals are enabled". Note that while I've given the generic case some thought, I've only implemented it for UARTs so far.

* All uart_* functions defined by the CPU's uart.o get renamed. (This can be done by a two-line `#define`/`#include`).
* The `uart_t` becomes a struct with a type and a union of what all used implementations use for their uart_t.
* A new module `mixed_uart.c` provides dispatcher functions.
* `UART_DEV(0)` keeps its semantics, but becomes a static inline function that returns a struct that, when dispatched, has the same effects as before. Other implementations would need to provide their own functions to that respect. (If `UART_DEV` is used anywhere to initialize a file-level uart_t, there might be need for a `UART_DEV_DECL(0)` that runs on preprocessor level. This is not a really new issue, just one that never surfaced because so far, no CPU uses `HAVE_UART_T` at all, and none decides it should be a struct, for which a shared initializer that works for function arguments and static-const initialization AFAICT can't be written in C).

This has the following desirable properties:

* No changes in preprocessed C code if it is not used, no changes to applications, minimal changes to UART implementations.
* In all places where the uart_t is passed to its functions directly (eg. `uart_write(STDIO_UART, ...)` where STDIO_UART is preprocessor-defined for `UART_DEV(0)`), a compiler with link time optimization should be able to emit the very instructions it would have emitted without mixed peripherals.

Changes I think are worth considering, but I wouldn't do them:

* As long as all `uart_t` fit in some limit (right now, it's all array indices), the mixed `uart_t` could be made an integer type that uses shifting or cast-to-struct to pack the type and the type-specific qualifier into a number. This would allow keeping `UART_DEV` in preprocessor. (Might be a worthwile optimization to save RAM, though).
* Parts of this are re-inventing classes / object oriented programming. C++ might have more ready-made solutions to this problem, but neither am I fluent with it, nor is it my impression that such an implementation would find wide acceptance in embedded programming.
* I did consider going through a per-implementation struct, so the dispatcher could be as generic as `void stdio_write(uart_t u, ...) { u.cls->write_fn(u, ...); }`, but that would
    a) require all implementations to agree on one inner uart_t if they were passed `u.arg` instead of `u`, or to dissect the mixed uart_t and pick their own data from the `.arg` union, and
    b) be hard to optimize even with link time optimization, because even if the class structs were `static const`, the compiler wouldn't be allowed to assume they'd always have the value they're initialized with.

Open questions:

* Both the mixed `uart_t` typedef and the dispatch function need to know all (possible / enabled) UART implementations. The union and function code could be easily populated from a list of them using X macros, but would require a single header file to be put together by the build system that enumerates them (eg. on an EFM32 build with SDO and pipe UART support, a `mixed_uart_implementations.h` would need to contain `#include <periph_cpu.h>` / `X(efm32)` / `#include <sdo_uart.h>` / `X(sdo)` / `#include <pipe_uart.h>` / `X(pipe)`). Is there a mechanism present in RIOT build infrastructure that can generate such a file?
* I feel like I'm polluting the name space by including lots of headers (the abovementioned) in `periph/uart.h`, where I need them to define the `uart_t.arg` union. Is that a problem at all, and if yes, any solutions?
* I'm not yet sure how in practice the other implementations' `UART_DEV()` functions would work. Right now, I'm using a `UART_PIPE(&my_uart_pipe)` style, but that requires an ugly `extern pipe_t my_uart_pipe` in periph/uart.h which is clearly not production style. The only way around I see is having the associated backends hard-enumerated by the respective UART implementations, allowing a `UART_PIPE(0)` to be used in any context -- but that seems to defeat the purpose of having a `HAVE_UART_T` mechanism in the first place. Maybe working on other implementations (like GPIO) will help with that.

I'll publish what I have so far in a branch and announce it here (although it's far from a pull request).

Would you consider a matured version of this for inclusion in RIOT?

gebart, 2016-06-13:
I would like to see the kind of high level abstraction/method dispatcher that you describe here, especially for the kind of devices where it's not uncommon to have multiple instances and different parts of the same category, such as UART (in-CPU, USB-CDC, debugger, etc.), flash memory (on-board memory, additional project-specific external memory), other non-volatile memory (eeproms etc.).

We have used a [similar solution] to what you describe in order to support write/read syscalls on mixed files and devices in the [Mulle port of Contiki]. In that solution we used structs with function pointers to mimic object oriented programming practices and let the syscalls dispatch the reads and writes to the functions pointed to by the struct. The solution does not need to know at compile time what types of devices there are but only that all of the drivers implement a common driver interface with function pointers. This kind of constructs are also used in Contiki for handling network drivers and REST server implementations. I guess this is one way to implement the `/dev/` nodes in Unix, I'm not familiar with the Linux kernel code to know if this is how it actually works though.

I am in favour of introducing a device dispatch in RIOT, but I think there might be some resistance from the low-end MCUs (avr, msp430), where RAM use for a few pointers may be a concern. I don't think the extra bytes or the extra code jump would matter on any of the Cortex-M3/M4's though, since the communication buses are usually orders of magnitude slower than RAM and core clocks.

I did not yet have the opportunity to look through your demo proposal which is referenced above, I will do so later when I have some spare time.

[similar solution]: http://www.billgatliff.com/newlib.html
[Mulle port of Contiki]: https://github.com/eistec/contiki/blob/mulle-master/cpu/arm/k60/include/devopttab.h

chrysn, 2016-06-15:
I've now uploaded a demo of this (applied to UARTs) to my [mixed-peripherals branch](https://github.com/chrysn-pull-requests/RIOT/tree/mixed-peripherals). The only CPU this was ported for now is EFM32 (which is currently out of tree) in [a dedicated branch](https://github.com/chrysn-pull-requests/EFM2Riot/tree/for-mixed-peripherals), but [the changes done to the CPU's native UARTs](https://github.com/chrysn-pull-requests/EFM2Riot/commit/2eec3fae539f64df36b944647f74e68b9b4fd636) should be rather simple to apply to any other UART implementation.

As indicated earlier, this is far from mergable. The biggest wart is that what should be done with X macros (the definition of the implementation enum, the individual unioned structs in the mixed uart_t, the switch cases) is still done manually; I hope there is a mechanism I'm not yet aware of in the build system that can be used to gather lines in the generated X macro header file, for that kind of things tends to be controversial when freshly introduced. (Of course we *could* do it without X macros at all and a registry of implemenations where each module can make its implementation available before the main invocation, making their function pointer struct known, but that would probably rule out any kind of compile time optimizability). Other warts are the uglinesses of the two current virtual UART implementations, and the uncertainty of how to properly initialize virtual UARTs: For hardwired ones, the board setup takes care of them and enumerates them, but the rest either needs static configuration somewhere (what is "Semihosted UART 0"?), possibly static allocation (right now I take the way of the boards, have the semihosted UART implementation allocate a fixed number of slots, and then whoever calls `SEMIHOSTED_UART_DEV(0, ...)` will need to coordinate with each other to not break each other). Special care has to be taken there because calls to `uart_init` can happen pretty early in the setup when used for stdio. (Ever tried to start a thread back then? Messy...).

The latest commit enables all of this in the default demo; the main function needs to explicitly start the semihosted implementation's listening thread -- both virtual UARTs are currently thread based because I did not yet get my debugger to send my MCU into a custom interrupt. The example works with an unmodified openocd by invoking `openocd -f board/efm32.cfg -c 'init; arm semihosting enable; monitor reset'`; your shell prompt is then directly at the openocd terminal. (You could just as well use a TTY or probably socket on the host side; ARM semihosting is used here, which allows the MCU to specify the desired filename to the host at socket setup time.)

Please have a look -- do you think this could (in a much refined state) be integrated into RIOT? Do you have suggestions on where in the make process to set the X macros? (It's still a month to go, but I'll be at the summit in Berlin; it'd be great if we could work out a better demonstration until then.)

chrysn, 2016-06-15:
About the openocd semihosted demo, I forgot (or initially thought we'd need that, tested it without, messed up the rebuild and thought it worked without): OpenOCD, following the blocking semantics of the ARM semihosting environment, blocks reads instead of reporting zero bytes read. Thus, when you run the example, RIOT spends all prompt waiting time waiting for the debugger to continue after the break point. To keep the system running, patch OpenOCD like this (until I've found a way to do select-style semantics with an interrupt to the MCU):

```
diff --git a/src/target/arm_semihosting.c b/src/target/arm_semihosting.c
index 21b7809..0b36bf4 100644
--- a/src/target/arm_semihosting.c
+++ b/src/target/arm_semihosting.c
@@ -98,6 +98,9 @@ static int do_semihosting(struct target *target)
 				if (strcmp((char *)fn, ":tt") == 0) {
-					if (m < 4)
+					if (m < 4) {
 						result = dup(STDIN_FILENO);
-					else
+						int flags = fcntl(result, F_GETFL, 0);
+						fcntl(result, F_SETFL, flags | O_NONBLOCK);
+					} else {
 						result = dup(STDOUT_FILENO);
+					}
 				} else {
```

chrysn, 2016-07-05:
Please have at least a rough look at the proposed code, I'd like to bring a slightly more mature version to the summit, but I'd need *some* feedback on whether such a path has a future in RIOT.

kaspar030, 2016-07-07:
While I totally agree that multi-API's are lacking in RIOT, I don't think this macro-based approach is the way to go. We try to avoid macro-magic wherever possible, for many reasons.

chrysn, 2016-07-07:
I'd like to mention that macros are used here only at two locations, neither of which is essential to the proposal:
* I've suggested that the pluggability of virtual devices be implemented using X macros. I'd be totally fine with any other form of code generation (which preprocessor basically is), but I don't see how an enum/switch can be extensible in C without generating code one way or another. (One could maybe do completely without if all dispatching were to be done via function pointer tables, but I think that wouldn't allow the compiler to do full optimization down to the same code as without abstraction).
* The `mask_uart.h` file is heavy preprocessor use, but that'd only serve as a way of not having to modify every single uart implementation. If it is preferable to actually rename every cpu's `uart_write` to `uart_write_${cpu}`, there wouldn't be a need for macros here.

kaspar030, 2016-07-08:
> I'd like to mention that macros are used here only at two locations,

Yes, but both are more than simple expressions. You will not find many
more throughout the RIOT code.

There are many clever ways of using macros in order to "extend" C in
desired ways, but so far we had a strong stance against that cleverness.

That is not set in stone, but will need a more general discussion (which
is out of scope for this PR).

closed 2019-09-10 by stale[bot]

chrysn, 2019-09-19:
The current approach to this is #9582, which solves the same problems and can supersede this issue.