RIOT-OS/RIOT #5506: How to access IEEE 802.15.4 MAC layer and change in MSBA2 Senor nodes

saleemrazam, 2016-06-03:
HI Oligham,

I am working on MSBA2 sensor, I am working on IEEE 802.15.4 MAC layer issues, Could you please guide me is there any way so that i can change MAC layer with my own developed MAC protocol. Could you please help me to know the step by step procedure to accomplish this task. Help would be highly appreciated

OlegHahm, 2016-06-03:
I guess you tried to address me (my name is Oleg Hahm).
1. Your question seems to be more appropriate for the mailing list.
2. I don't see why you're addressing particularly me.
3. 802.15.4 won't work on the MSBA2 since the TI CC1100 doesn't support 802.15.4 PHY.
4. Have you looked at the existing PRs concerning MAC layer implementations?

PeterKietzmann, 2017-01-13:
Question was answered and no response since half a year.

closed 2017-01-13 by PeterKietzmann