RIOT-OS/RIOT #5512: tests: printf_float reports false negative

miri64, 2016-06-04:
On native:

```
Value displayed: 2016
[FAILED] Values are not equal:
actual: 2016
expected: 2016.0
```

gebart, 2016-06-05:
I get the same error on both native and on Mulle with newlib (version from newlib Git around january 2016).
The test is wrong, native is correct, according to documentation on the standard printf:

http://pubs.opengroup.org/onlinepubs/009695399/functions/fprintf.html:
> An optional precision that gives the minimum number of digits to appear for the d, i, o, u, x, and X conversion specifiers; the number of digits to appear after the radix character for the a, A, e, E, f, and F conversion specifiers; the maximum number of significant digits for the g and G conversion specifiers; or the maximum number of bytes to be printed from a string in the s [XSI] [Option Start]  and S [Option End]  conversion specifiers. **The precision takes the form of a period ( '.' ) followed either by an asterisk ( '*' ), described below, or an optional decimal digit string, where a null digit string is treated as zero.** If a precision appears with any other conversion specifier, the behavior is undefined.

A-Paul, 2016-06-08:
This issue has been addressed already by https://github.com/RIOT-OS/RIOT/pull/5383. Followed by an extended approach in https://github.com/RIOT-OS/RIOT/pull/5384.

miri64, 2016-06-08:
Ah, okay then let's close this as duplicate.

A-Paul, 2016-06-09:
Reopen because although there has been PRs fixing this error, no issue existed before. So this issue isn't a duplicate.
This issue can give some motivation to get https://github.com/RIOT-OS/RIOT/pull/5383 and maybe https://github.com/RIOT-OS/RIOT/pull/5384 merged. ;-)

kYc0o, 2016-07-11:
Since #5383 was merged, can we reclose this issue?

miri64, 2016-07-11:
If the test is working now, yes.

kaspar030, 2016-07-11:
works for me.

closed 2016-07-11 by kaspar030