RIOT-OS/RIOT #5520: tests/periph_uart not working

kYc0o, 2016-06-06:
I'm struggling a lot with this test. Even for samr21-xpro the second UART seems not to work, however I think it's due to the [ringbuffer](https://github.com/RIOT-OS/RIOT/blob/master/tests/periph_uart/main.c#L70) which somehow retains the data.

Anyone can test if it's my setup or there is a bug on this?

I'm testing with a [FTDI TTL-232R-3V3](http://www.ftdichip.com/Support/Documents/DataSheets/Cables/DS_TTL-232R_CABLES.pdf) cable.

kYc0o, 2016-06-06:
For info, the test works well if RX and TX are interconnected directly.

kaspar030, 2016-06-06:
I suppose you connect the ground lines?

kYc0o, 2016-06-06:
Yes of course

kYc0o, 2016-06-06:
Another hint: if I change the default STDIO output to UART1 for instance, I can have the shell on an external terminal client using the same cable...

Cr0s, 2016-08-06:
This test is awaits a trailing zero character at the end of a transmission. Do you tried to explicitly send a zero character?

PeterKietzmann, 2016-08-09:
Tried with samr21-xpro and a CP210x UART/USB converter. Pyterm didn't show any output, Minicom did.

kYc0o, 2016-08-09:
I'm on OSX using CoolTerm btw, but no output as well...

gebart, 2016-08-09:
I didn't look at the test, but some general information that might be relevant here: Pyterm doesn't print anything until a newline is received. Flow control (cts, dtr) should be off. Did you compare if there are any differences between the settings in the two programs?

kYc0o, 2017-01-09:
So actually the test was fine but required a specific character to print the characters. So this is closed thanks to #6302

closed 2017-01-09 by kYc0o