RIOT-OS/RIOT #5530: gnrc: support sending data frames with zero-length payloads

aeneby, 2016-06-08:
Currently it is not possible to send data frames with zero-length payloads via the GNRC stack, due to the check [here](https://github.com/RIOT-OS/RIOT/blob/e9bbd54f5825c6fddedf2030ecddae7a5703cbc1/sys/net/gnrc/pktbuf_static/gnrc_pktbuf_static.c#L97). Data frames with zero-length payloads are still valid 802.15.4 packets (PAN coordinators are required to be able to send them in response to data request commands, for example), and therefore RIOT's network stack should support sending them.

miri64, 2016-06-08:
GNRC supports sending of zero-length data frames: it uses NULL pointers for that (see e. g. ping6 command).

aeneby, 2016-06-08:
Okay, perhaps I misunderstood something - I did eventually manage to send a zero-length payload directly from a test application. My initial test case however was via the shell:

```
txtsnd 4 bcast ""
2016-06-08 20:14:18,300 - INFO # > txtsnd 4 bcast ""
2016-06-08 20:14:18,302 - INFO # error: packet buffer full
```

I would have expected that to send out a zero-length packet.

aeneby, 2016-06-08:
So if I understand you correctly, what you're saying is that the bug is actually in `_netif_send()` rather than `gnrc_pktbuf_add()`, since [here](https://github.com/RIOT-OS/RIOT/blob/e9bbd54f5825c6fddedf2030ecddae7a5703cbc1/sys/shell/commands/sc_netif.c#L1076) we should be checking if the data is zero-length, and if so, creating a packet with a NULL data pointer?

miri64, 2016-06-08:
Yes :)

aeneby, 2016-06-08:
Okay, I've changed the issue's title and description to reflect the actual problem. Will submit a new PR.

PR #5534 (2016-06-09): shell: set NULL pointer for data pakets with zero-length payload
This PR makes it possible to send empty packets using `txtsnd` at the shell, e.g.:

`txtsnd 4 bcast ""`

Previously `_netif_send()` would attempt to create an empty packet for the NETIF header to point to, however `gnrc_pktbuf_add()` won't allow snips with zero length. Now, the NETIF header pointer is set to NULL for zero-length snips instead.

Fixes #5530.
closed 2016-06-09 by miri64