RIOT-OS/RIOT #5535: gnrc: network stack does not handle data packets with zero-length payload

aeneby, 2016-06-09:
This bug is easily reproducible using the native port and PR #5534 (or just master once the PR is merged). Simply set up a two-node topology as described [here](https://github.com/RIOT-OS/RIOT/wiki/Virtual-riot-network), and send an empty data packet from one node to the other:

`txtsnd 4 11:22:33:44:55:66 ""`

The receiving node will promptly dump core - I believe the crash is due to a [failed assertion](https://github.com/RIOT-OS/RIOT/blob/29f0eea2efe39be1384cfb71a839ede2e7583b17/sys/net/gnrc/pktbuf_static/gnrc_pktbuf_static.c#L224) but I don't yet understand why it fails.

miri64, 2016-06-09:
Using gdb and `assert.h` I traced it down to the packet parsing in `sys/net/gnrc/link_layer/netdev2/gnrc_netdev2_eth.c` for native. Will provide a fix for all link-layer types.

miri64, 2016-06-09:
See #5536

miri64, 2016-06-09:
Related issue #5530.

aeneby, 2016-07-21:
ACK frames will also provoke this.

miri64, 2016-09-28:
#5694 was merged.

closed 2016-09-28 by miri64