RIOT-OS/RIOT #5538: Ncache lost after reboot, other nodes won't be in cache anymore

bapclenet, 2016-06-10:
My setup:
- Border router on samr21
- Gnrc_networking node A
- Gnrc_networking node B

Border router powers up, then node A and B. Node A and B are in ncache of BR.
Then BR reboots, ncache is erased. node A and B keeps their global addresses but they are never added in ncache of BR.

Why the nodes does not remove their global address or why ncache is not filled again?

miri64, 2016-06-10:
> Why the nodes does not remove their global address?

Why should they or to ask it more direct: how should they know they should? They got a PIO with a prefix and a valid lifetime for it, now they get a new PIO with (another?) prefix and a valid lifetime for it. Unless the old prefix times out or the router explicitly sends a 0-lifetime PIO it will not be deleted.

> Why ncache is not filled again?

Here's how it should happen:
1. 6LBR sends out a RA
2. 6LN receives it configures new address and solicits address registration (send NS with ARO, get NA from 6LBR back with ARO signalling success).

Did any of that happen in your set-up? (Just checking, that the ND is broken is a known issue, that I plan to fix the next weeks).

miri64, 2016-06-10:
Normally after a reboot an updated version ABRO should be send, since the version is not stored persistently this most likely does not happen => 6LN recognizes update as update => doesn't re-register.

bapclenet, 2016-06-13:
>Normally after a reboot an updated version ABRO should be send, since the version is not stored persistently this most likely does not happen => 6LN recognizes update as update => doesn't re-register.

Thanks for your answer. 
Is it possible to control when 6LBR sends out a RA? I would like to send it just during a slot of time (5 min) then stop. Is that even possible or 6LN will remove is global address if it doesn't reecive a new RA so its prefix has not a valid lifetime?
Basically, I would like to say from now to now+5min, send RA so nodes are added to network then stop sending RA and no new nodes will be accepted in the network.

miri64, 2016-06-13:
> Is it possible to control when 6LBR sends out a RA?

Usually that interval is "randomized" between `GNRC_IPV6_NETIF_DEFAULT_MIN_ADV_INT` and `GNRC_IPV6_NETIF_DEFAULT_MAX_ADV_INT` (but given the fact that we still don't initialize a sensible seed for our PRNG its always the same). If you change that you can manipulate it. You can also just unset the `RTR_ADV` flag on your interface using `ifconfig <if> -rtr_adv` or `gnrc_ipv6_netif_set_rtr_adv(ip6_netif, false)` (from `net/gnrc/ipv6/netif.h`) after 5 min.

> . Is that even possible or 6LN will remove is global address if it doesn't reecive a new RA so its prefix has not a valid lifetime?

Since currently GUAs are not removed when their lifetime runs out this won't happen. In fact the [send-out of periodic RA isn't required in 6LN](https://tools.ietf.org/html/rfc6775#section-6.4), so it should work (not when unsetting the flag though, since then it won't reply to RS by the nodes either)

bapclenet, 2016-06-14:
I see, I'll make some try later.

> Here's how it should happen:
1. 6LBR sends out a RA
2. 6LN receives it configures new address and solicits address registration (send NS with ARO, get NA from 6LBR back with ARO signalling success).
Did any of that happen in your set-up? (Just checking, that the ND is broken is a known issue, that I plan to fix the next weeks).

On which file I should enable debug to see this message? So I'll be able to dig more on what happe and what I expect.

Let's close this issue since it is a normal behaviour.

bapclenet, 2016-06-21:
ping @miri64

bapclenet, 2016-06-29:
ping @miri64

miri64, 2016-07-04:
Sorry, was on vacation. Will look into it ASAP.

miri64, 2016-07-04:
> On which file I should enable debug to see this message? So I'll be able to dig more on what happe and what I expect.

Best thing would be to use a sniffer I think (e.g. in the IoT-Lab testbed). Alternatively, you can enable debug in

* `sys/net/gnrc/network_layer/ndp/gnrc_ndp.c`
* `sys/net/gnrc/network_layer/ndp/internal/gnrc_ndp_internal.c`
* `sys/net/gnrc/network_layer/sixlowpan/nd/gnrc_sixlowpan_nd.c`

and to look out for the DEBUG output of the `..._handle()` functions for each node.

bapclenet, 2016-07-04:
Thanks, let's close this PR and I'll come back with more detail if necessary

closed 2016-07-04 by bapclenet