RIOT-OS/RIOT #5539: Border Router: uhcp

bapclenet, 2016-06-10:
My setup:

Linux with start_network.sh (i.e ethos and uhcpd running)
Border router on samr21 
Gnrc_networking node A
Gnrc_networking node B

I run a program that sends UDP packets to my nodes every 20 sec.
After some time, instead of seeing:
got packet from fe80::46:a5ff:fe3d:fc3e port 58493
uhcp: push from fe80::46:a5ff:fe3d:fc3e:58493 prefix=2001:db8::/64
gnrc_uhcpc: uhcp_handle_prefix(): got same prefix again

uhcp client gives:
uhcp_client(): timeout waiting for reply
uhcp_client(): sending REQ...
...
In a loop. 

I thought it was uhcpd which was buggy so I restarted it, it works but still no communication between BR and Linux. So I try to ping from BR to Linux ping6 fe80::1%7, no chance (same in the other way)
What's the problem, why can't Linux communicate to BR anymore? 

If I reboot the BR, it works and I see again that uhcp client and server can communicates.
How can I debug it more?

miri64, 2016-07-05:
So the gist is, somehow everything is 0 if I see that correctly

bapclenet, 2016-07-05:
Yes, I only keep ping6 from linux, it works for a while and then it does not work anymore, I see that packets get stopped with ``gnrc_netdev2: unable to forward packet of type 0``

How can I debug more? Can the gnrc_pktsnip_t be corrupted somewhere?

bapclenet, 2016-07-05:
What could cause the packet to be malformed every time?

smlng, 2016-07-06:
could this be related to (or even fixed by) #5596?

bapclenet, 2016-07-06:
@smlng, I don't think so, because packet is locked in netdev2 so before ipv6 layer if I'm right?

bapclenet, 2016-07-06:
@miri64, any idea on that?

kaspar030, 2016-07-06:
> What could cause the packet to be malformed every time?

It is possible that ethos' state somehow get's corrupted.

bapclenet, 2016-07-06:
@kaspar030, ok, how can I check that? 
I can easily reproduce it by keeping a ping on 2001:db8:: node or even interface 6 on BR

Here is a gist with some more debugs:
https://gist.github.com/bapclenet/2fa3c4185633d827980109aab0003644

bapclenet, 2016-07-06:
@kaspar030, on ethos, is there some controls that a packet is correctly arrived on BR before sending a new one?

bapclenet, 2016-07-06:
I tried at 9600Baud and with 48MHz or 8Mhz, it does not change, pkt gets corrupted at some point.

Did someone try to continuously ping a node from linux (with Global IP)?

bapclenet, 2016-07-06:
In the irq_handler of UART, I checked if I have an error:
```
static inline void irq_handler(int dev)
{
    SercomUsart *uart = _uart(dev);

    if (uart->STATUS.reg & SERCOM_USART_STATUS_BUFOVF) {
        printf("SERCOM_USART_STATUS_BUFOVF\n");
        uart->STATUS.reg = SERCOM_USART_STATUS_BUFOVF;
    }
```

And yes BUFOVF is raised each time before it starts to not working

bapclenet, 2016-07-06:
I think, I understand the bug, when I constantly ping a node, it sends a ping every 1sec. UHCP sends a request every 1min and if by bad luck uhcp server sends answer at the same time as my ping, Linux being much quicker than samr21, linux sends both message very quickly and RIOT does not have time to deal with them. 
It is an assumption, but adding a little delay in the loop in linux ethos helps to not have collision. 
The ideal solution would be to have a queue in ethos linux and before sending a new message it should get an ACK from ethos RIOT.
@kaspar, did you think about that when you wrote ethos?

bapclenet, 2016-07-08:
Tell me if I'm wrong but the packet sent by serial to netdev is allocated somewhere or it is still allocated in the ringbuffer of ethos? If it is kept in the ringbuffer and a new packet arrives before the other has finished to be transmitted, the packet is overwritten? @kaspar030

kYc0o, 2016-07-26:
Anyone against to postpone this bug to be solved on the next release?

kYc0o, 2016-07-26:
Postponed.

biboc, 2016-08-26:
@kaspar030, have you tried to reproduce the bug?

miri64, 2016-10-17:
@kaspar030 ping?

miri64, 2016-11-08:
This sounds a little like the analysis given by @Yonezawa-T2 in #5926. @biboc can you check if #6064 fixes this issue for you?

biboc, 2016-11-09:
I don't think this issue is related because this is more about ethos communication over serial.

biboc, 2016-11-09:
Did you try to send many packets in a row to see if the communication is still on?

kYc0o, 2017-04-26:
#6064 was merged, @biboc can you test if this is still an issue?

biboc, 2017-04-26:
I'm sorry @kYc0o , I won't be able to test this for now. 
I don't think #6064 is related to this bug.
Currently, with a small sleep (50ms) before sending a packet from ethos to BR, there are no problem.
This problem may exist due to hardware configuration and not ethos software so if nobody encounters it, I would vote to close this issue.

miri64, 2017-11-06:
Fixed by #7925

miri64, 2017-11-27:
@biboc can you test if this is still an issue, now that #7925 is merged?

biboc, 2017-11-28:
I have no time now to test it unfortunately
This issue is very old and IMO should be removed

miri64, 2017-11-28:
I'll try to see what I can do, once I'm through with testing for #5790.

miri64, 2017-11-28:
I actually think this was covered in my earlier tests for #7925, unless I'm missing something. But since you @biboc as OP also say it should be closed, I close it now (untested, but pretty confident that it is fixed).

closed 2017-11-28 by miri64