RIOT-OS/RIOT #5541: SAMR21 Xplained - Hardfault

gevard, 2016-06-11:
Hello

I'm completly newby to riot-os, I found it with an excellent article in a french magazine call "Open Silicium". 

After doing some exercise like helloword and gnrc_networking, I tried the example of the article, with is related to 6LowPan.  
I got an hardfault, it's reproducible, and occurs at the server side when I launch the device. Before that the server stay alive and tell me the it open the UDP connection.
But as soon, I launch the device (client with sensor), the server has that Hardfault.

Any help is welcome.

[Open Silicium reference link](http://www.opensilicium.com/?p=98#more-98)
[Github page of the article ](https://github.com/aabadie/riot-apps/tree/master/point_to_point_network)

### Flashing Target ###
Open On-Chip Debugger 0.10.0-dev-00322-g406f4d1 (2016-06-04-14:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
none separate
adapter speed: 400 kHz
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.1A.00FB
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 400 kHz
Info : SWD DPIDR 0x0bc11477
Info : at91samr21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* at91samr21g18.cpu  cortex_m   little at91samr21g18.cpu  halted
at91samr21g18.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000418 msp: 0x20004280
auto erase enabled
Info : SAMD MCU: SAMR21G18A (256KB Flash, 32KB RAM)
wrote 49152 bytes from file /home/gevard/riot-os/Exercice_OpenSilicium/6LowPan/Server/bin/samr21-xpro/server_to_terminal.hex in 10.768572s (4.457 KiB/s)
at91samr21g18.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000418 msp: 0x20004280
verified 39208 bytes in 3.243132s (11.806 KiB/s)
shutdown command invoked
Done flashing
/home/gevard/riot-os/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM1" -b "115200"
No handlers could be found for logger "root"
2016-06-11 11:02:45,146 - INFO # Connect to serial port /dev/ttyACM1
Welcome to pyterm!
Type '/exit' to exit.
2016-06-11 11:02:49,249 - INFO # �����jFK�B�I���
Context before hardfault:
2016-06-11 11:02:49,251 - INFO #    r0: 0x00000000
2016-06-11 11:02:49,252 - INFO #    r1: 0x00000000
2016-06-11 11:02:49,254 - INFO #    r2: 0x2000008c
2016-06-11 11:02:49,255 - INFO #    r3: 0x00010a00
2016-06-11 11:02:49,257 - INFO #   r12: 0x20000d80
2016-06-11 11:02:49,259 - INFO #    lr: 0x000018c3
2016-06-11 11:02:49,260 - INFO #    pc: 0x000018ca
2016-06-11 11:02:49,262 - INFO #   psr: 0x01000000
2016-06-11 11:02:49,262 - INFO # 
2016-06-11 11:02:49,262 - INFO # Misc
2016-06-11 11:02:49,264 - INFO # EXC_RET: 0xfffffffd
2016-06-11 11:02:49,268 - INFO # Attempting to reconstruct state for debugging...
2016-06-11 11:02:49,269 - INFO # In GDB:
2016-06-11 11:02:49,270 - INFO #   set $pc=0x18ca
2016-06-11 11:02:49,271 - INFO #   frame 0
2016-06-11 11:02:49,272 - INFO #   bt
Traceback (most recent call last):
  File "/home/gevard/riot-os/RIOT/dist/tools/pyterm/pyterm", line 738, in <module>
    myshell.cmdloop("Welcome to pyterm!\nType '/exit' to exit.")
  File "/usr/lib/python2.7/cmd.py", line 130, in cmdloop
    line = raw_input(self.prompt)
KeyboardInterrupt
/home/gevard/riot-os/RIOT//Makefile.include:333: recipe for target 'term' failed
make: *** [term] Interrupt

aabadie, 2016-06-13:
Hi @gevard.
Thanks for reporting, there's indeed an issue with the networking example when they are built against the master branch.
You can try with version 2016.04 and it works. In the RIOT source directory, use:
```
$ git checkout 2016.04
```

I'll dig a bit more into this in a few days to make the examples compatible with latest master. @miri64, any suggestion on how to fix it ?

miri64, 2016-06-13:
I would suggest to git-bisect it from last week, since it was working then,this way we might find the culprit that broke it. Will do this afternoon.

miri64, 2016-06-13:
BTW: I would suggest to use `2016.04-branch` instead of `2016.04` to get the newest fixes ;-)

miri64, 2016-06-13:
Anyway, I just tried the example you provided, but I wasn't able to reproduce... But maybe I'm doing something wrong... @aabadie can you try to bisect when the crash was introduced?

aabadie, 2016-06-13:
By default, I didn't have the crash either. But simply adding `printf("message: %s", message)` in `server_thread` right the call to `snprintf` triggered the crash. Otherwise nothing happened.

Try the server_to_terminal example and send udp packets on port 8000 to it.

----- Mail original -----

gevard, 2016-06-18:
Your expectation is completely correct, I added a printf for debugging purpose, and now doing _$ git checkout 2016.04_ it working as expected. Thank you very much for helping.

aabadie, 2016-06-20:
> Thank you very much for helping.

You're welcome.
Thanks for reporting BTW, you highlighted issues with the UDP layers in the current code.

PeterKietzmann, 2016-06-20:
Is the issue addressed? In that case, please close it @aabadie.

aabadie, 2016-06-20:
Issue addressed. Closing.

closed 2016-06-20 by aabadie