RIOT-OS/RIOT #5542: gnrc: consider removing restriction on registering NETTYPE_NETIF packets

aeneby, 2016-06-11:
Currently it's not possible to register packets of this type, due to restrictions imposed by a [check](https://github.com/RIOT-OS/RIOT/blob/789b2c0917507058fe10bf312c36ca34284e4fa6/sys/net/gnrc/netreg/gnrc_netreg.c#L28) in `gnrc_nettype_register()`. If this is a legacy restriction (i.e. one imposed prior to the existence of NETTYPE_NETIF), then perhaps it can be removed or modified to permit registration of this type.

Such a change would allow, for example, data packets with zero-length payloads, and acknowledgement response packets, to be examined using tools such as `pktdump`.

miri64, 2016-06-11:
I'm still not 100% sure what the point of this is. Handling such packets usually is something the MAC layer should do and it has full access to it. If you want to examine packets why not use the sniffer application, which is much more comfortable to use, since you can use wireshark with it.

aeneby, 2016-06-11:
My understanding is that the purpose of `pktdump` is primarily as a debug tool, and the fact that it is ignoring (or cannot display) certain types of perfectly valid 802.15.4 packets is a bug, IMO, even if alternatives exist. Perhaps the proposed solution is the wrong one - perhaps the packets in question should be getting marked as `NETTYPE_UNDEF` instead, for example - but I thought I would at least present the argument for discussion.

I don't agree that this is restricted to the MAC layer, as these types of packets should be propagating all the way through the stack.

aeneby, 2016-06-16:
To simplify the question - is there currently any way for a RIOT application to know that a zero-length data packet has been received?

aeneby, 2016-06-30:
> is there currently any way for a RIOT application to know that a zero-length data packet has been received?

@miri64 ?

miri64, 2016-07-04:
The "problem" is a more a matter of memory optimization: NETIF is intentionally not >=0 because every `nettype` >=0 consumes memory in `netreg`. It's just a word-length per type but just for debugging purposes I don't see why we should waste them. We could make NETIF >=0, if DEVELHELP is defined, but that would make the code unnecessary complicated in that case.

> To simplify the question - is there currently any way for a RIOT application to know that a zero-length data packet has been received?

Of what layer? Because for e.g. UDP I definitely would be a bug. For any other (except NETIF): yes by registering for the type of the header.

Come to think of it: I think in the end it might even be cleaner to allow for 0-length pktsnips with NULL pointers on the data. This way registration for the zero-length *payload* would be possible and some weird registration requirements for the 0-length payloads with regards to the headers wouldn't be necessary.

miri64, 2016-10-18:
Is this still needed, now that #5694 was merged?

closed 2016-10-18 by miri64