RIOT-OS/RIOT #5543: stm32: compiler errors in DAC

OlegHahm, 2016-06-11:
Trying to compile any application for iotlab-m3 on current master results in:
```
"make" -C /home/oleg/git/RIOT/cpu/stm32_common/periph
dac.c: In function 'dac_set'
dac.c:76:19: error: array subscript is above array bounds [-Werror=array-bounds]
     if (dac_config[line].chan) {
         ~~~~~~~~~~^~~~~~
dac.c: In function 'dac_poweron'
dac.c:91:38: error: array subscript is above array bounds [-Werror=array-bounds]
     DAC->CR |= (1 << (16 * dac_config[line].chan));
                            ~~~~~~~~~~^~~~~~
dac.c: In function 'dac_poweroff'
dac.c:96:39: error: array subscript is above array bounds [-Werror=array-bounds]
     DAC->CR &= ~(1 << (16 * dac_config[line].chan));
                             ~~~~~~~
```

I assume that the cause for this error exists for some time, but shows only now because of an update gcc:
```
arm-none-eabi-gcc (Arch Repository) 6.1.1 20160526
```

OlegHahm, 2016-06-11:
A simple fix might be:
```diff
diff --git a/cpu/stm32_common/periph/dac.c b/cpu/stm32_common/periph/dac.c
index d250802..79c6464 100644
--- a/cpu/stm32_common/periph/dac.c
+++ b/cpu/stm32_common/periph/dac.c
@@ -71,6 +71,10 @@ int8_t dac_init(dac_t line)
 
 void dac_set(dac_t line, uint16_t value)
 {
+#ifndef DAC_CONFIG
+    (void) line;
+    (void) value;
+#else
     value = (value >> 4);       /* scale to 12-bit */
 #ifdef DAC_DHR12R2_DACC2DHR
     if (dac_config[line].chan) {
@@ -84,11 +88,16 @@ void dac_set(dac_t line, uint16_t value)
 
     _DAC(line)->DHR12R1 = value;
 #endif
+#endif
 }
 
 void dac_poweron(dac_t line)
 {
+#ifndef DAC_CONFIG
+    (void) line;
+#else
     DAC->CR |= (1 << (16 * dac_config[line].chan));
+#endif
 }
```

miri64, 2016-06-12:
Duplicate / subpart of https://github.com/RIOT-OS/RIOT/issues/5519

closed 2016-06-12 by miri64

OlegHahm, 2016-06-12:
Ups, overlooked this one.