RIOT-OS/RIOT #5554: Add PWM driver for NRF5x

sdalu, 2016-06-15:
nRF52 natively support PWM driver, and it's possible to implement it for nRF51:
 https://github.com/NordicSemiconductor/nrf51-pwm-library

PeterKietzmann, 2016-06-16:
Hi @sdalu what do you mean by "native PWM driver support"? Usually in RIOT we write drivers ourselves to have a slim implementation that matches our hardware abstraction and API which is not dependant on externals libs... Did I miss the point?

sdalu, 2016-06-16:
Sorry wrong wording I guess. nRF52 has a PWM module builtin, so this chip will only require a slim implementation. nRF51 lack such PWM module but this can be achieved using TIMER, GPIOTE and PPI modules

PeterKietzmann, 2016-06-16:
So the proposed library is a software PWM implementation that uses standard I/O pins and timers to generate a PWM modulated signal?

sdalu, 2016-06-16:
Yes, the provided link is the Nordic software implementation of PWM for the nRF51.

PeterKietzmann, 2016-06-16:
@sdalu I appreciate your efforts to improve nRF support in RIOT. But as said, using external librarys for peripheral drivers is not the "RIOT way". Looking at the implementation you posted and at the reference manual for nRF51 controllers, it does not seem too complex to implement a driver by ourselves. Now the questions are (i) do you urgently *need* this feature for some project and (ii) would you volunteer to provide a RIOT driver? Didn't check the license but probably one could reuse some code then.  In any case, I could support with testing on the nrf51dongle as well as the yunjia board.

sdalu, 2016-06-16:
I provided the link for reference implementation purpose
(i) that's a feature that i will need
(ii) i can volunteer to test

PeterKietzmann, 2016-06-16:
Okay. So then we still need someone who actually writes the driver :-). I won't find time for this. Any volunteers? I will hold my eyes and ears open...

PeterKietzmann, 2018-11-09:
Nowadays both nrf51 an nrf52 implement PWM drivers.

closed 2018-11-09 by PeterKietzmann